mpromonet/v4l2rtspserver #172: Recommendations for Better HLS Performance Pi0W

drewcovi, 2020-01-24:
Hello there! Amazing package here. Love the results.

Any recommendations on how I could further optimize my setup?

I'm able to stream the `rtsp` stream smoothly consuming only 25% of my CPU and maybe 42 megs of memory total.

**But it chokes on HLS.** :(

I've disabled `Power Management` on wlan0

```uv4l --driver raspicam --auto-video_nr --vflip --framerate 10 --width 540 --height 960 --hflip --encoding h264

v4l2rtspserver /dev/video0,default -F 10 -G 544x960x10 -C 1 -S
```

mpromonet, 2020-01-25:
Hi,

Good that you find this piece of software interesting.

What means chockes ? 100% of CPU ? crashed ?
Maybe you can make some tries with `bcm2835-v4l2` driver.

Best Regards,
Michel.

drewcovi, 2020-01-27:
For some reason HLS simply doesn't stream effectively. It starts and then stops routinely

drewcovi, 2020-01-27:
I'll give that other driver a try :)

mpromonet, 2020-03-04:
Hi,
At least this one is opensource, and then you may understand problems reading the code.
Best Regards,
Michel.

closed 2020-04-12 by mpromonet