mpromonet/v4l2rtspserver #173: Multiple clients connecting to server

Synoptic2, 2020-02-06:
I am trying to get two client (for example, 2 VLC client) connect to the same v4l2rtspserver.
It looks like they are battling with the connection. Sometimes they both get the stream, but most of the time, one gets it, and the other doesn't.

Est-ce que v4l2rtspserver supporte plusieurs clients qui se connectent sur la même instance ?

Merci,

mpromonet, 2020-02-13:
Hi @Synoptic2,

I don't see any reason why connecting multiple clients raise a problems, except if it overflow some resource capacity like networks, cpu, ...

Did you try with smaller definition ?

Best Regards,
Michel.

Synoptic2, 2020-02-13:
I think it may simply overflow the capacity. I did not try with a smaller definition either.

closed 2020-04-12 by mpromonet