mpromonet/v4l2rtspserver #176: Can't start v4l2rtspserver in docker because of: mmap: No such device

konstantinj, 2020-02-11:
Hi,

I have basically the same issue as in #153 but I'd like to use the docker version on my pi3 with the picam.

uv4l is installed successfully but I can't figure out what I need to mount to make it work inside the docker container. This at least is not working:

`sudo uv4l --driver raspicam --device-name=video6 --encoding h264`

```
docker run --name v4l2rtspserver --restart always --device=/dev/video6 -p 8554:8554 \
-v/usr/lib/uv4l:/usr/lib/uv4l \
-v/etc/uv4l:/etc/uv4l \
-v/usr/bin/uv4l:/usr/bin/uv4l \
-eLD_PRELOAD=/usr/lib/uv4l/uv4lext/armv6l/libuv4lext.so \
-it mpromonet/v4l2rtspserver -s -W1280 -H720 /dev/video6
```

konstantinj, 2020-02-11:
Just figured out it works when using the `-r` option. Would still like to have it without though.

mpromonet, 2020-02-13:
Hi @konstantinj 

I guess you should inherit the docker installing uv4l inside.
But why not just using the driver `bcm2835-v4l2` that works well with raspicam ?

Best Regards,
Michel.

konstantinj, 2020-02-14:
You're right! When I was reading the readme I did not get that somehow. Now it's working, thanks.

closed 2020-02-14 by konstantinj