mpromonet/v4l2rtspserver #177: How do I use the -f arg?

comrade1, 2020-02-14:
My usb camera supports mjpg and yuyv. mjpg is the default.

`decker@gr1:~ $ v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
	[1]: 'YUYV' (YUYV 4:2:2)
[video4linux2,v4l2 @ 0x105d1c0] Compressed:       mjpeg :          Motion-JPEG : 2592x1944 1280x960 1024x768 640x480 1280x720 1600x1200 2048x1536 800x600 1920x1080
[video4linux2,v4l2 @ 0x105d1c0] Raw       :     yuyv422 :           YUYV 4:2:2 : 2592x1944 1280x960 1024x768 640x480 1280x720 1600x1200 2048x1536 800x600 1920x1080`

My software that connects to the rtsp unicast drops the connection frequently and the monitoring software presents the message 'unable to determine image size'. (vlc seems to hold the connection fine though)
In the logs I see:
`[NOTICE] /home/decker/tmp/v4l2rtspserver/src/DeviceSource.cpp:131
	V4L2DeviceSource::doStopGettingFrames`

So, I was thinking I would try outputting YUYV from the camera and using v4l2compress_h264 to send h264 to my monitoring software.

I've tried -fyuyv, -fraw, -fyuyv422

I see something like:
`[NOTICE] /home/decker/tmp/v4l2rtspserver/src/main.cpp:564
	Version: 0.1.9-5-g72f6d1a live555 version:2020.02.11
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/main.cpp:608
	Create V4L2 Source.../dev/video0
[NOTICE] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
	/dev/video0 support capture
[NOTICE] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:130
	/dev/video0 support streaming
[NOTICE] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:62
	/dev/video0:MJPG size:1920x1080 bufferSize:4147200
[ERROR] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:193
	Cannot set pixelformat to:yuyv format is:MJPG
[ERROR] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:74
	Cannot init device:/dev/video0
[NOTICE] /home/decker/tmp/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
	Device /dev/video0
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/main.cpp:661
	Create ALSA Source...plughw:2,0
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/ALSACapture.cpp:49
	Open ALSA device: "plughw:2,0"
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/ALSACapture.cpp:117
	set sample format device: plughw:2,0 to:2 ok
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/ALSACapture.cpp:104
	ALSA device: "plughw:2,0" buffer_size:524288 period_size:512 rate:48000
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread
[NOTICE] /home/decker/tmp/v4l2rtspserver/src/main.cpp:150`

so, it starts out trying to create a V4L2 source, but then you see:
`	Cannot set pixelformat to:yuyv format is:MJPG`

Is there a way to make the camera output YUYV, which it says that it supports?

comrade1, 2020-02-14:
I think I figured it out. It had nothing to do with the -f argument. Instead:

> v4l2-ctl --set-fmt-video=width=1920,height=1080,pixelformat=YUYV
> v4l2compress_h264 /dev/video0 /dev/video100 &
> v4l2rtspserver -a S16_LE -C 2 -A 48000 -H 1080 -W 1920 -F 15 -P 8555 /dev/video0,plughw:2,0
>

mpromonet, 2020-02-14:
Hi comrade1,
In order to set format to YUYV, you should simply use `-fYUYV` (using the FOURCC).
As I told before, I am not sure the video/RAW format is correctly implemented (maybe some conversion are missing).
Best Regards,
Michel.

closed 2020-04-12 by mpromonet