mpromonet/v4l2rtspserver #178: Save to file -O option creates an empty file

YoavGit, 2020-02-21:
Hi Michel, 
When i add the -O <filename> flag it creates an empty file with no record of the video.

What am i doing wrong? 
Thanks.

mpromonet, 2020-02-26:
Hi dekashio,
`-O` is something I did for debug, maybe you should add `-w` to make it work with a file (by default it is using mmap interface that works with an output v4l2 device)
Best Regards,
Michel.

dekashio, 2020-02-26:
Hi Michel, thanks for the reply, I tried also running: v4l2rtspserver -w test.mp4 /dev/video15 and still getting 0 byte file. 
not sure what i'm missing here.
I'd like to be able to stream and save locally the video from the device.
thanks in advance.

mpromonet, 2020-02-26:
Hi,
With -w the command should looks like `v4l2rtspserver -w -O test.h264 /dev/video15`.
The output will not be muxed as an mp4, it should be what was captured (with some SPS/PPS repeat).
Best Regards,
Michel.

dekashio, 2020-02-27:
that worked! thanks alot :)

closed 2020-02-29 by mpromonet