mpromonet/v4l2rtspserver #182: ZoneMinder Error - Invalid SDP format at 'video3/unicast'	zm_rtsp.cpp	380

webgps, 2020-04-20:
Hey there again,

Trying to add a stream to ZM.
I'm getting the following error:
zmc_m2		5477	ERR	Invalid SDP format at 'video3/unicast'	zm_rtsp.cpp	380

My zm source config is:

[image]

Please help.

Thanks in andvance!

mpromonet, 2020-04-20:
Hi webgps,

This depends on how you start v4l2rtspserver and the parameter you gave.
If the url is correct, this probably means zoneminder doesnot support the capture format you are using ?

Best Regards,
Michel.

webgps, 2020-04-20:
This is my command:

./v4l2rtspserver  /dev/video0 /dev/video1 /dev/video3 -H640 -W480 -f

mpromonet, 2020-04-20:
Hi,

Then the rtsp url are /video0/unicast /video1/unicast /video3/unicast
You can get the SDP using openRTSP to see what it looks like, maybe this will helps you to understand if it is supproted by zoneminder.

Best Regards,
Michel.

webgps, 2020-04-22:
So if I play the strem with openRTSP i will see the *.sdp string at he end of the url or something like that?

mpromonet, 2020-04-27:
Hi @webgps,

Using `openRTSP rtsp:// ..../video3/unicast` it will print the SDP string in the answer of RTSP DESCRIBE request.

Best Regards,
Michel.

programminghoch10, 2020-07-16:
Just use ffmpeg or libvlc to capture the rtsp stream!

closed 2020-12-17 by mpromonet