mpromonet/v4l2rtspserver #185: Problems starting the v4l2rtspserver service with UV4L drivers

jarrah31, 2020-04-29:
Hi Michel,

First let me say thanks for writing a very useful utility! I'm using it (or trying to use it) for RTSP streams from Raspberry Pi's to video surveillance software SecuritySpy. When it works the lag is sub one second which is fantastic!

The problem I have is with starting the v4l2rtspserver service to use the UV4L video stream. It basically goes into a crash loop and has to be stopped. However, manually starting v4l2rtspserver outside of the service works fine. I can repeat the problem on different fresh installs of Raspbian Buster, and with Pi4 and 3B+.

The main reason for trying to use UV4L is so that I can rotate my camera and overlay a realtime clock (which works surprisingly well), otherwise I'd just use v4l2rtspserver.

Here are my service files:
/etc/systemd/system/uv4l_raspicam.service
```
[Unit]
Description=UV4L Raspicam

[Service]
Type=simple
ExecStart=/usr/bin/uv4l -f -k --sched-fifo --mem-lock --config-file=/etc/uv4l/uv4l-raspicam.conf --driver raspicam --driver-config-file=/etc/uv4l/uv4l-raspicam.conf --enable-server off
Restart=on-abnormal
LimitNOFILE=65536

[Install]
WantedBy=multi-user.target
```

/etc/systemd/system/v4l2rtspserver.service
```
[Unit]
Description=V4L2 RTSP server
After=network.target

[Service]
Type=simple
Restart=always
RestartSec=1
ExecStart=/usr/local/bin/v4l2rtspserver -U user:pass -F10 -W 1296 -H 972 -P 8555 /dev/video1
User=pi
StartLimitIntervalSec=0

[Install]
WantedBy=multi-user.target
```

These are the different scenarios I've tried:
1. v4l2rtspserver.service only using open source drivers on video0. Service starts ok.
2. uv4l_raspicam.service and v4l2rtspserver.service started. v4l2rtspserver keeps crashing.
3. Manually running the uv4l and v4l2rtspserver commands (as they are shown within ExecStart). Both work fine, v4l2 doesn't crash.
4. uv4l_raspicam.service started and v4l2rtspserver manually run, both work fine. 
5. uv4l manual start, v4l2rtspserver.service. v4l2rtspserver enters a crash loop.

The output of the crash loop can be seen here:
```
$ sudo systemctl start v4l2rtspserver; journalctl -u v4l2rtspserver -f
-- Logs begin at Wed 2020-04-29 10:54:39 BST. --
Apr 29 11:06:03 testcam systemd[1]: /etc/systemd/system/v4l2rtspserver.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
Apr 29 11:06:03 testcam systemd[1]: Started V4L2 RTSP server.
Apr 29 11:06:03 testcam v4l2rtspserver[596]: log level:500
Apr 29 11:06:04 testcam v4l2rtspserver[596]: mmap: No such device
Apr 29 11:06:04 testcam v4l2rtspserver[596]: mmap: No such device
Apr 29 11:06:04 testcam v4l2rtspserver[596]: mmap: No such device
Apr 29 11:06:04 testcam v4l2rtspserver[596]: munmap: Invalid argument
Apr 29 11:06:04 testcam v4l2rtspserver[596]: munmap: Invalid argument
Apr 29 11:06:04 testcam v4l2rtspserver[596]: [NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:564
Apr 29 11:06:04 testcam v4l2rtspserver[596]:         Version: 0.2.0-6-g3542b6f live555 version:2020.04.24
Apr 29 11:06:04 testcam v4l2rtspserver[596]: [NOTICE] /home/pi/v4l2rtspserver/src/main.cpp:608
Apr 29 11:06:04 testcam v4l2rtspserver[596]:         Create V4L2 Source.../dev/video1
Apr 29 11:06:04 testcam v4l2rtspserver[596]: [NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:124
Apr 29 11:06:04 testcam v4l2rtspserver[596]:         driver:raspicam capabilities:5000001 mandatory:4000001
Apr 29 11:06:04 testcam v4l2rtspserver[596]: [NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:127
Apr 29 11:06:04 testcam v4l2rtspserver[596]:         /dev/video1 support capture
Apr 29 11:06:04 testcam v4l2rtspserver[596]: [NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:129
[... 118 lines cut ...]
Any help you can give would be much appreciated as I've been going round in circles trying to get this to work. :)

This is my installation and setup recipe if it helps.
```
curl http://www.linux-projects.org/listing/uv4l_repo/lpkey.asc | sudo apt-key add -
sudo vim /etc/apt/sources.list
	deb http://www.linux-projects.org/listing/uv4l_repo/raspbian/stretch stretch main
sudo apt update
sudo apt install cmake liblog4cpp5-dev libv4l-dev git vim uv4l uv4l-server uv4l-raspicam uv4l-raspicam-extras
sudo vim /etc/uv4l/uv4l-raspicam.conf
	encoding = h264
	width = 1296
	height = 972
	framerate = 10
	exposure = auto
	rotation = 180
cd
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver/
cmake .
make
sudo make install
```

Thanks.

mpromonet, 2020-04-29:
Hi jarrah31,

As uv4l is closed-source it is difficult to know what is going wrong, I completly stop to use it since bcm2835-v4l2 is working well.
Anyway uv4l need to preload library to override the v4l2 calls, you probaly need to configure `LD_PRELOAD` to the path of `libuv4lext.so`

Best Regards,
Michel.

jarrah31, 2020-04-29:
Is there a way to rotate the camera using the open source drivers? If there is I'm happy to ditch UV4L. :)

jarrah31, 2020-04-29:
aha, found this:
sudo v4l2-ctl -c rotate=180
What would be the best way to integrate this into the service ExecStart command? (I'll try a few things in the meantime)

jarrah31, 2020-04-29:
Right, just in case it helps others, I've put it in /etc/rc.local where it works nicely.

```
# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi
v4l2-ctl -c rotate=180
exit 0
```

closed 2020-04-29 by jarrah31