mpromonet/v4l2rtspserver #187: URL Traversal Issue w/Live555 Proxy

CharlesCage, 2020-05-07:
Not entirely sure is this is just my own configuration stupidty of an issue, but I discovered that on the Pi where I have this deployed I can access etc/passwd by traversing via the streaming URL. Specifically:

`http://[camera IP]:8554/../../../../../../etc/passwd`

A GET request returns the file.

Appreciate any help you could offer I've looked for a place where I can configure the LIVE555 proxy to prevent this, but can't find one. Is it something I can configure?

mpromonet, 2020-05-08:
Hi CharlesCage,

The embeded HTTP server avoid navigation with `..` (HTTPServer.cpp)[https://github.com/mpromonet/v4l2rtspserver/blob/master/src/HTTPServer.cpp]. 
Do you access to /etc/passwd using v4l2rtspserver or another software ?

Best Regards,
Michel.

CharlesCage, 2020-05-14:
I'm accessing via v4l2rtpserver.

mpromonet, 2020-05-14:
Dear CharlesCage,

I guess you are using a webroot that allow you to access to /etc, If you start v4l2rtspserver with `-b/` or `-b/etc` obviously you can access to /etc/passwd.
By default the webroot is the working directory, so if you start `v4l2rtspserver` from `/` you expose the whole filesystem.

Best Regards,
Michel.

CharlesCage, 2020-05-14:
I'm starting v4l2rtspserver with the following:

`/usr/local/bin/v4l2rtspserver -F 30 -W 1280 -H 720`

Also confirmed that I'm _not_ running the service in which this starts as root.

mpromonet, 2020-05-14:
Hi @CharlesCage,

Then this depends what is the current directory when you run the command `/usr/local/bin/v4l2rtspserver -F 30 -W 1280 -H 720`
Adding `-b/tmp` should fix your problem, or maybe better using a unexisting path ?

Best Regards,
Michel.

CharlesCage, 2020-05-15:
Thank you! This solved my issue. Appreciate your patience, and of course all your work on this project. :)

closed 2020-05-15 by CharlesCage

mpromonet, 2020-05-15:
You are welcome.

Maybe I will change this, the http server is a kind of hack of RTSP server in order to serve the HLS/MPEG# segments, maybe it will be a good thing to use civetweb as http server instead of this.