mpromonet/v4l2rtspserver #190: ALSA device is available but cannot use ALSA command

imchesterhk, 2020-05-18:
Hi,

Thank you for the great work, i am able to get the video from the USB HDMI capture card and stream it via network, but seem i cannot get the audio . and i checked with the command with `arecord -l`, shows below.

`pi@raspberrypi:~/v4l2rtspserver $  arecord -l`
`**** List of CAPTURE Hardware Devices ****`
`card 2: Device_1 [Live Streaming USB Device], device 0: USB Audio [USB Audio]`
`  Subdevices: 1/1`
`  Subdevice #0: subdevice #0`

but when i try with command
`v4l2rtspserver -W 1920 -H 1080 -F 29 -p 8555 /dev/video0,hw:2 -A 48000 -a S16_LE `

it just shows the full help page and it is not showing the ALSA part. Is it something wrong with my hardware thats not support with this v4l2rtspserver ?  Thank you ..

mpromonet, 2020-05-18:
Hi imchesterhk,

If the ALSA part is not shown using `-h` argument, it means it has been built without ALSA support.
When cmake run it detects if ALSA is available and print `Alsa available`, if not ALSA is disabled (and then giving `-a` `-A` or `-C` will return the help).   
Installing ALSA dev package and rebuilding should solve the problem.

Best Regards,
Michel.

imchesterhk, 2020-05-18:
if i do the command `cat /proc/asound/cards`
it shows
`pi@raspberrypi:~/v4l2rtspserver $ cat /proc/asound/cards`
` 0 [ALSA           ]: bcm2835_alsa - bcm2835 ALSA`
`                      bcm2835 ALSA`
` 2 [Device_1       ]: USB-Audio - Live Streaming USB Device`
`                      eEver Live Streaming USB Device at usb-0000:01:00.0-2, super speed`
does it really mean my device is not support ?
and can you more specific about "Installing ALSA dev package and rebuilding" ?
is it i need to install some new package to solve this problem ? or have to rebuild some software by myself to solve this ? i am new about this, sorry for bother .. and thank you so much !

mpromonet, 2020-05-18:
Hi,

If you are using the arm release from https://github.com/mpromonet/v4l2rtspserver/releases, the package `libasound2-dev` is not installed. The dev package is need to brings libraries and include used by ALSA support.

Using the snap release https://build.snapcraft.io/user/mpromonet/v4l2rtspserver it is built with sound support.   

When I was saying "Installing ALSA dev package", this could be done running: 
```
sudo apt install  g++ make cmake pkg-config liblog4cpp5-dev libasound2-dev
git clone https://github.com/mpromonet/v4l2rtspserver
cd v4l2rtspserver
cmake . && make && make install
``` 

Best Regards,
Michel.

imchesterhk, 2020-05-18:
```
sudo apt install  g++ make cmake pkg-config liblog4cpp5-dev libasound2-dev
git clone https://github.com/mpromonet/v4l2rtspserver
cd v4l2rtspserver
cmake . && make && make install
```

After i doing this, i am now able to stream the video and audio together !!!
Thank you again Michel !

closed 2020-05-18 by imchesterhk