mpromonet/v4l2rtspserver #192: Can't get video to flip...

earlthesquirrel, 2020-05-30:
I need to flip video (because of how raspberry Pi zero is set up in the mounting I have) but
v4l2-ctl --set-ctrl=vertical_flip=1,horizontal_flip=1

doesn't stick. (or seem to have any impact)

Can I set this on the server line: v4l2rtspserver  -H 680 -W 420 -F 15 -P 8555 /dev/video0
as an argument?  Or is there something else that might be wrong?

Also tried : 
v4l2-ctl --set-ctrl vertical_flip=1
 v4l2-ctl --set-ctrl horizontal_flip=1

No errors and no impact...

stefan-sherwood, 2020-07-14:
`v4l2-ctl --set-ctrl=horizontal_flip=1` flips the image for me on my Rpi Zero W, even dynamically while v4l2rtspserver is already running. I am using a stock Rpi camera rev 1.3.

mpromonet, 2020-09-19:
I close this issue, probably linked to the V4L2 device used, this project just read V4L2 frame and expose it as an RTSP server.

closed 2020-09-19 by mpromonet