mpromonet/v4l2rtspserver #196: udp or rtp connection

rayzie, 2020-07-09:
Hello.
I am now using RTSP Server with your help to provide the best streaming service.
There is one thing to worry about.
Now I'm using the server in the following way:

`sudo ./h264_v4l2_rtspserver/v4l2rtspserver -F 25 -W 1280 -H 720 -P 8555 /dev/video0`

I would like to change to UDP or RTP method for the above server.
Please tell me how to structure the Script.

And what about the url for connecting to a UDP streaming server?

Thank you.

mpromonet, 2020-09-19:
Hi rayzie,

There is no implementation of UDP streaming, you can use RTP streaming without RTSP layer using multicast option `-M`.
By default it choose multicast group & ports, you can also specify them using `-M234.1.2.3:50000` next you will get the RTP stream on port 50000 of 234.1.2.3 multicast group.

Best Regards,
Michel.

wvdakker, 2022-05-11:
Hi, I am facing the same thing.
I got the multicast running. However how do I run my client?
I use 'mpv rtp://239.0.0.5:5000 --rtsp-transport=udp_multicast' but no stream is shown.
Also tried with '/multicast', with or whitout --rtsp-transport.
Tried ffprobe, ffplay. Same thing, no input is shown.

I use ip route add 239.0.0.0/24 dev <interface>'
Further more tried an sdp file, no results.

Is there any working client with config for multicasts?

mpromonet, 2022-05-11:
Hi wvdakker,

I don't see what is similar to the initial issue... did you try `mpv rtsp://<server_ip>/multicast --rtsp-transport=udp_multicast`  ?

Best Regards,
Michel.

wvdakker, 2022-05-11:
Yes tried that also:
[ffmpeg] tcp: Connection to tcp://192.168..x.x failed Connection refused.

I am using: v4l2rtspserver -W 1024 -H 576 -F 8 -M 239.0.0.5:20005 /dev/video0
In my (perhaps wrong) opinion I should use mpv rtp://239.0.0.5:5000 --rtsp-transport=udp_multicast.
Or do I make a mistake in thinking .....

mpromonet, 2022-05-11:
Hi,

I don't know how mpv wrap ffmpeg option (however rtsp-transport option has probably no impact using an rtp url).
It is working with : 
```
v4l2rtspserver -M239.0.0.5:20005 &
ffplay -fflags nobuffer rtp://239.0.0.5:20005
```

Best Regards,
Michel.

wvdakker, 2022-05-11:
If I use your ffplay command I get
 Unable to receive RTP payload type 96 without an SDP file describing it
rtp://239.0.0.5:20005: Invalid data found when processing input

Googling for SDP file says that a RTP stream cannot be shown because
the stream has no payload and the player dont know what to use.

closed 2023-04-29 by mpromonet