mpromonet/v4l2rtspserver #197: v4l2rtspserver can't grab any frames on Raspberry Pi Zero W after upgrading from 4.19.118+ #1311 to 5.4.51+ #1327

danieltwagner, 2020-07-27:
I had a v4l2rtspserver working nicely on my Raspberry Pi Zero W but after I updated it stopped working.

Kernel that worked:
```
Linux pi-camera 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
```

Kernel that doesn't work:
```
Linux pi-camera 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
```

```
$ v4l2rtspserver -W 1640 -H 1232 -F 25 -P 8555 -S /dev/video0 -vv
2020-07-27 19:38:46,652 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
	level:DEBUG
2020-07-27 19:38:46,660 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:385
	Version: 0.2.0-19-g0fa3368 live555 version:2020.07.21
2020-07-27 19:38:46,673 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:426
	Create V4L2 Source.../dev/video0
2020-07-27 19:38:46,680 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:bm2835 mmal capabilities:85200005 mandatory:4000001
2020-07-27 19:38:46,683 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video0 support capture
2020-07-27 19:38:46,686 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
	/dev/video0 support read/write
2020-07-27 19:38:46,688 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video0 support streaming
2020-07-27 19:38:46,690 [DEBUG ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:71
	/dev/video0:H264 size:1640x1232 bufferSize:2050048
2020-07-27 19:38:46,698 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:215
	/dev/video0:H264 size:1640x1232 bufferSize:2050048
2020-07-27 19:38:46,701 [DEBUG ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:71
	/dev/video0:H264 size:1640x1232 bufferSize:2050048
2020-07-27 19:38:46,705 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:236
	fps:1/25
2020-07-27 19:38:46,707 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:237
	nbBuffer:1
2020-07-27 19:38:46,709 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
2020-07-27 19:38:46,743 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
2020-07-27 19:38:46,750 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:0 size:2050048 offset:0
2020-07-27 19:38:46,752 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:1 size:2050048 offset:2052096
2020-07-27 19:38:46,756 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:2 size:2050048 offset:4104192
2020-07-27 19:38:46,756 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:3 size:2050048 offset:6156288
2020-07-27 19:38:46,759 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:4 size:2050048 offset:8208384
2020-07-27 19:38:46,764 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:5 size:2050048 offset:10260480
2020-07-27 19:38:46,768 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:6 size:2050048 offset:12312576
2020-07-27 19:38:46,770 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:7 size:2050048 offset:14364672
2020-07-27 19:38:46,771 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:8 size:2050048 offset:16416768
2020-07-27 19:38:46,774 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:9 size:2050048 offset:18468864
2020-07-27 19:38:47,173 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:449
	Create Source .../dev/video0
2020-07-27 19:38:47,178 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:140
	Queue is empty
2020-07-27 19:38:47,182 [NOTICE] - /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:82
	Play this stream using the URL "rtsp://10.0.102.237:8555/ts"
2020-07-27 19:38:47,186 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:527
	HLS       http://10.0.102.237:8555/ts.m3u8
2020-07-27 19:38:47,187 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:528
	MPEG-DASH http://10.0.102.237:8555/ts.mpd
2020-07-27 19:38:47,188 [NOTICE] - /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:82
	Play this stream using the URL "rtsp://10.0.102.237:8555/unicast"
2020-07-27 19:38:47,190 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
	begin thread

[connect client]

RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!

[nothing else appears]

^CSIGINT
2020-07-27 19:39:18,721 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:549
	Exiting....
```

I'm thinking the `totSessionBW` (might be a live 555 thing?) might be a red herring and may have been there before even while things were working. As you can see there is virtually no output despite `-vv`. All I did was on a clean Raspberry Pi OS Buster checkout perform an `apt upgrade`.

The following both work nicely after the upgrade:
```
$ raspistill -o out.jpg
$ raspivid -t 0 --mode 4 -w 1640 -h 1232 --codec H264 -b 2500000 --profile high --level 4.2 --irefresh adaptive --inline -l -o tcp://0.0.0.0:3333 -v -a 127
```
The latter is a (poor) replacement for the functionality that I'm missing with v4l2rtspserver not working, as I can access the stream from VLC with `tcp/h264://<ip>:3333`

danieltwagner, 2020-07-27:
I found this resolved issue: https://github.com/mpromonet/v4l2rtspserver/issues/162

Neither `-r` nor `-B` or the combination of both appears to make any difference.

```
$ strace v4l2rtspserver -W 1640 -H 1232 -F 25 -P 8555 -S /dev/video0 -vv -r -B  2>&1 | grep -v gettimeofday

[... the process is now waiting for a connection]

_newselect(4, [3], [], [], {tv_sec=0, tv_usec=9472}) = 0 (Timeout)
_newselect(4, [3], [], [], {tv_sec=0, tv_usec=9482}) = 0 (Timeout)
_newselect(4, [3], [], [], {tv_sec=0, tv_usec=9476}) = 1 (in [3], left {tv_sec=0, tv_usec=8669})
accept(3, {sa_family=AF_INET, sin_port=htons(51463), sin_addr=inet_addr("10.0.102.130")}, [16]) = 5
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6bb3120}, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6bb3120}, 8) = 0
fcntl64(5, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(5, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
getsockopt(5, SOL_SOCKET, SO_SNDBUF, [102400], [4]) = 0
getsockopt(5, SOL_SOCKET, SO_SNDBUF, [102400], [4]) = 0
brk(0x8fc000)                           = 0x8fc000
_newselect(6, [3 5], [], [5], {tv_sec=0, tv_usec=7743}) = 1 (in [5], left {tv_sec=0, tv_usec=7695})
recvfrom(5, "OPTIONS rtsp://pi-camera"..., 20000, 0, 0xbeacfe8c, [16->0]) = 133
send(5, "RTSP/1.0 200 OK\r\nCSeq: 2\r\nDate: "..., 152, 0) = 152
_newselect(6, [3 5], [], [5], {tv_sec=0, tv_usec=9479}) = 1 (in [5], left {tv_sec=0, tv_usec=1024})
recvfrom(5, "DESCRIBE rtsp://pi-camera"..., 20000, 0, 0xbeacfe8c, [16->0]) = 159
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_IP) = 6
setsockopt(6, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
setsockopt(6, SOL_SOCKET, SO_REUSEPORT, [1], 4) = 0
setsockopt(6, SOL_IP, IP_MULTICAST_LOOP, "\1", 1) = 0
fcntl64(6, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(6, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
mmap2(NULL, 2052096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb5c0b000
munmap(0xb5c0b000, 2052096)             = 0
close(6)                                = 0
getsockname(5, {sa_family=AF_INET, sin_port=htons(8555), sin_addr=inet_addr("10.0.102.237")}, [16]) = 0
send(5, "RTSP/1.0 200 OK\r\nCSeq: 3\r\nDate: "..., 631, 0) = 631
_newselect(6, [3 5], [], [5], {tv_sec=0, tv_usec=9641}) = 1 (in [5], left {tv_sec=0, tv_usec=3748})
recvfrom(5, "SETUP rtsp://10.0.102.237:8555/u"..., 20000, 0, 0xbeacfe8c, [16->0]) = 183
getsockname(5, {sa_family=AF_INET, sin_port=htons(8555), sin_addr=inet_addr("10.0.102.237")}, [16]) = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_IP) = 6
setsockopt(6, SOL_SOCKET, SO_REUSEADDR, [0], 4) = 0
setsockopt(6, SOL_SOCKET, SO_REUSEPORT, [0], 4) = 0
setsockopt(6, SOL_IP, IP_MULTICAST_LOOP, "\1", 1) = 0
bind(6, {sa_family=AF_INET, sin_port=htons(6970), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
socket(AF_INET, SOCK_DGRAM|SOCK_CLOEXEC, IPPROTO_IP) = 7
setsockopt(7, SOL_SOCKET, SO_REUSEADDR, [0], 4) = 0
setsockopt(7, SOL_SOCKET, SO_REUSEPORT, [0], 4) = 0
setsockopt(7, SOL_IP, IP_MULTICAST_LOOP, "\1", 1) = 0
bind(7, {sa_family=AF_INET, sin_port=htons(6971), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
fcntl64(6, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(6, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
mmap2(NULL, 2052096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb5c0b000
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
getsockopt(6, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
send(5, "RTSP/1.0 200 OK\r\nCSeq: 4\r\nDate: "..., 214, 0) = 214
_newselect(6, [3 5], [], [5], {tv_sec=0, tv_usec=8935}) = 1 (in [5], left {tv_sec=0, tv_usec=2411})
recvfrom(5, "PLAY rtsp://10.0.102.237:8555/un"..., 20000, 0, 0xbeacfe8c, [16->0]) = 162
getsockname(5, {sa_family=AF_INET, sin_port=htons(8555), sin_addr=inet_addr("10.0.102.237")}, [16]) = 0
fcntl64(7, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(7, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
getsockopt(7, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
getsockopt(7, SOL_SOCKET, SO_SNDBUF, [180224], [4]) = 0
write(2, "RTCPInstance::RTCPInstance error"..., 77RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
) = 77
setsockopt(7, SOL_IP, IP_MULTICAST_TTL, "\377", 1) = 0
sendto(7, "\200\310\0\6\17K\247E\342\311\241W\317\215\35\212\227\216\255b\0\0\0\0\0\0\0\0\201\312\0\7"..., 60, 0, {sa_family=AF_INET, sin_port=htons(61459), sin_addr=inet_addr("10.0.102.130")}, 16) = 60
getsockname(7, {sa_family=AF_INET, sin_port=htons(6971), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
mmap2(NULL, 2052096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb5a16000
send(5, "RTSP/1.0 200 OK\r\nCSeq: 5\r\nDate: "..., 187, 0) = 187
_newselect(8, [3 5 7], [], [5], {tv_sec=0, tv_usec=9607}) = 1 (in [7], left {tv_sec=0, tv_usec=9566})
recvfrom(7, "\316\372\355\376", 1422, 0, {sa_family=AF_INET, sin_port=htons(61459), sin_addr=inet_addr("10.0.102.130")}, [16]) = 4
_newselect(8, [3 5 7], [], [5], {tv_sec=0, tv_usec=1753}) = 1 (in [7], left {tv_sec=0, tv_usec=1712})
recvfrom(7, "\316\372\355\376", 1422, 0, {sa_family=AF_INET, sin_port=htons(61459), sin_addr=inet_addr("10.0.102.130")}, [16]) = 4
_newselect(8, [3 5 7], [], [5], {tv_sec=0, tv_usec=9646}) = 0 (Timeout)
_newselect(8, [3 5 7], [], [5], {tv_sec=0, tv_usec=9643}) = 0 (Timeout)
_newselect(8, [3 5 7], [], [5], {tv_sec=0, tv_usec=9655}) = 0 (Timeout)

[keeps repeating]
```

mpromonet, 2020-07-27:
Hi,

In order to analyze strace output you should either add `-f` option to strace in order to follow threads or add `-s` option to use only main thread.
I made a quick try with kernel 5.4.51-v7+ on a raspberry pi2 but did not have the problem you describe.

Best Regards,
Michel.

danieltwagner, 2020-07-27:
Thanks for your prompt response, Michael. I'm not sure if that sheds a whole lot more light but for completeness I'm including the top of the main thread as well:

```
$ strace -o out -fff v4l2rtspserver -W 1640 -H 1232 -F 25 -P 8555 -S /dev/video0 -vv -r -B  2>&1 | grep -v gettimeofday
```

out.828
```
execve("/usr/local/bin/v4l2rtspserver", ["v4l2rtspserver", "-W", "1640", "-H", "1232", "-F", "25", "-P", "8555", "-S", "/dev/video0", "-vv", "-r", "-B"], 0xbeb5b6e0 /* 20 vars */) = 0
brk(NULL)                               = 0xe91000
uname({sysname="Linux", nodename="pi-camera.de.dtw.sh", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f35000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/local/bin/v4l2rtspserver", 4096) = 29
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\250\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
fstat64(3, {st_mode=S_IFREG|0644, st_size=9512, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f33000
mmap2(NULL, 73772, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ef3000
mprotect(0xb6ef5000, 61440, PROT_NONE)  = 0
mmap2(0xb6f04000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6f04000
close(3)                                = 0
munmap(0xb6f35000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=28908, ...}) = 0
mmap2(NULL, 28908, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f2b000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0(\20\1\0004\0\0\0"..., 512) = 512
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
[... 436 lines cut ...]
close(3)                                = 0
close(1)                                = 0
write(1, "SIGINT\n", 7)                 = -1 EBADF (Bad file descriptor)
exit_group(0)                           = ?
+++ exited with 0 +++
```

out.829
```
set_robust_list(0xb6b1d4c0, 12)         = 0
mmap2(NULL, 2097152, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_NORESERVE, -1, 0) = 0xb5d02000
munmap(0xb5d02000, 1040384)             = 0
munmap(0xb5f00000, 8192)                = 0
mprotect(0xb5e00000, 135168, PROT_READ|PROT_WRITE) = 0
gettimeofday({tv_sec=1595879552, tv_usec=153269}, NULL) = 0
write(1, "2020-07-27 20:52:32,153 [NOTICE]"..., 97) = 97
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0} <unfinished ...>) = ?
+++ exited with 0 +++

```

danieltwagner, 2020-07-27:
Thanks for your prompt response, Michael. I'm not sure if that sheds a whole lot more light but for completeness I'm including the top of the main thread as well:

```
$ strace -o out -fff v4l2rtspserver -W 1640 -H 1232 -F 25 -P 8555 -S /dev/video0 -vv -r -B  2>&1 | grep -v gettimeofday
```

out.828
```
execve("/usr/local/bin/v4l2rtspserver", ["v4l2rtspserver", "-W", "1640", "-H", "1232", "-F", "25", "-P", "8555", "-S", "/dev/video0", "-vv", "-r", "-B"], 0xbeb5b6e0 /* 20 vars */) = 0
brk(NULL)                               = 0xe91000
uname({sysname="Linux", nodename="pi-camera", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f35000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/local/bin/v4l2rtspserver", 4096) = 29
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\250\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
fstat64(3, {st_mode=S_IFREG|0644, st_size=9512, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f33000
mmap2(NULL, 73772, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ef3000
mprotect(0xb6ef5000, 61440, PROT_NONE)  = 0
mmap2(0xb6f04000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6f04000
close(3)                                = 0
munmap(0xb6f35000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=28908, ...}) = 0
mmap2(NULL, 28908, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f2b000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0(\20\1\0004\0\0\0"..., 512) = 512
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
_llseek(3, 206384, [206384], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 206740, [206740], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
[... 436 lines cut ...]
close(3)                                = 0
close(1)                                = 0
write(1, "SIGINT\n", 7)                 = -1 EBADF (Bad file descriptor)
exit_group(0)                           = ?
+++ exited with 0 +++
```

out.829
```
set_robust_list(0xb6b1d4c0, 12)         = 0
mmap2(NULL, 2097152, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_NORESERVE, -1, 0) = 0xb5d02000
munmap(0xb5d02000, 1040384)             = 0
munmap(0xb5f00000, 8192)                = 0
mprotect(0xb5e00000, 135168, PROT_READ|PROT_WRITE) = 0
gettimeofday({tv_sec=1595879552, tv_usec=153269}, NULL) = 0
write(1, "2020-07-27 20:52:32,153 [NOTICE]"..., 97) = 97
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
_newselect(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0} <unfinished ...>) = ?
+++ exited with 0 +++

```

danieltwagner, 2020-08-02:
I was able to narrow it down to GPU memory consumption, but I'm not able to get it to work at the desired 1640x1232 (2x2 binning on Camera Module v2):
This works as already mentioned above:
```
raspivid -t 0 --mode 4 -w 1640 -h 1232 --codec H264 -b 2500000 --profile high --level 4.2 --irefresh adaptive --inline -l -o tcp://0.0.0.0:3333 -v -a 127
```

... but ffmpeg fails with a somewhat useful error code:
```
$ ffmpeg -f v4l2 -input_format h264 -pix_fmt yuyv422 -video_size 1640x1232 -framerate 25 -i /dev/video0 -c:v copy -pix_fmt yuyv422 output.mkv
[...]
[video4linux2,v4l2 @ 0x189fec0] ioctl(VIDIOC_STREAMON): Operation not permitted
/dev/video0: Operation not permitted
```

It turns out that we fail to allocate some memory:
```
$ vcgencmd mem_reloc_stats
alloc failures:     1
compactions:        0
legacy block fails: 0
```

My `/etc/config.txt` contains `gpu_mem=128` and stats look like so when the pi is idle:
```
$ vcgencmd get_mem arm && vcgencmd get_mem gpu && vcgencmd get_mem malloc_total && vcgencmd get_mem malloc && vcgencmd get_mem reloc_total && vcgencmd get_mem reloc
arm=384M
gpu=128M
malloc_total=10M
malloc=8M
reloc_total=108M
reloc=106M
```

Running at a lower resolution, both ffmpeg and v4l2rtspserver work just fine:
```
$ v4l2rtspserver -W 1280 -H 720 -F 25 -P 8555 -S /dev/video0 -v
[...]
2020-08-02 15:02:11,015 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	intv_sec:1596376931 fps:25 bandwidth:2453kbps
2020-08-02 15:02:11,022 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
	outtv_sec:1596376931 fps:25 bandwidth:2453kbps
```

Memory looks happy:
```
$ vcgencmd get_mem arm && vcgencmd get_mem gpu && vcgencmd get_mem malloc_total && vcgencmd get_mem malloc && vcgencmd get_mem reloc_total && vcgencmd get_mem reloc
arm=384M
gpu=128M
malloc_total=10M
malloc=5M
reloc_total=108M
reloc=48M
```

Increasing `gpu_mem` to 192 or 256 doesn't make 2x2-binned 1640x1232 work, but ffmpeg now has a different complaint:
```
[video4linux2,v4l2 @ 0x103eec0] Dequeued v4l2 buffer contains 4100096 bytes, but 4040960 were expected. Flags: 0x00002001.
/dev/video0: Invalid data found when processing input
```

At this point I feel like this is nothing to do with your software anymore and that I might have to take this up with the raspberry pi folks. What do you think? I assume you have not been able to repro?

mpromonet, 2020-08-02:
Hi danieltwagner,

If I follow you, after an upgrade of kernel capturing frame of 1280x720 doesnot work anymore, but it works with lower resolution ?
I made a try with the kernel version you mentionned but not with this resolution.   
If ffmeg can capture using the V4L2driver, it may have a problem in the capture code...

Best Regards,
Michel.

danieltwagner, 2020-08-02:
Sorry, let me clarify:
After the kernel upgrade:
- raspivid can capture 1640x1232
- ffmpeg can capture up to 1280x720 but fails for 1640x922 or 1640x1232 (maybe due to gpu memory, and with more `gpu_mem` buffer sizes mismatch)
- v4l2rtspserver can capture up to 1640x922, but fails for 1640x1232 (even with more `gpu_mem`)

azsde, 2020-12-15:
I do have the same issue regarding the max resolution: v4l2rtspserver can capture up to 1640x922, but fails for 1640x1232

WiredLife, 2021-02-13:
Has anyone a solution? Have the same problem with newest kernel @ raspbian.

mystcb, 2021-11-01:
Just wanted to add a little more to this, as I have had the same issue. I have been watching this issue since I had an identical issue to @danieltwagner (down to the same logs so no point in posting them).

I have been lucky enough to get one of the new RPi Zero 2 W's and ran the same configuration on this (both using a similar image, Ansible to deploy the same code etc), and get the same problem. Including being unable to downgrade the Kernel to restore the higher resolution camera image.

Not sure what else I can add, other than having tried all of the options above and ran into the same issues, esp on the new Pi's as well.

zeretar, 2021-11-01:
I hit and fixed a similar issue over the weekend also with the Zero 2 W and RPi Camera v2.1. While tracking it down I noticed that the bcm2835_v4l2 kernel module was defaulting to a max_video_width/height of 1280x768 for some reason. Changing these to the max of that module (3280x2464) seemed to resolve it for me.

_Checking the current module parameters:_
`$ tail -n +1 /sys/module/bcm2835_v4l2/parameters/*`

_Setting more accurate defaults_
```
$ sudo nano /etc/modprobe.d/bcm2835_v4l2.conf
options bcm2835_v4l2 max_video_height=2464
options bcm2835_v4l2 max_video_width=3280

$ sudo update-initramfs
```

mystcb, 2021-11-01:
I will continue trying, but following the above returns the error of: 

`VIDIOC_STREAMON: Invalid argument`

when using the command:

`/usr/local/bin/v4l2rtspserver -W 1640 -H 1232 -P 8554 /dev/video0`

Having double-checked the settings, I can see that the module has had its parameters updated: 

<details>
<summary>Output from parameter check</summary>

```
# tail -n +1 /sys/module/bcm2835_v4l2/parameters/*
==> /sys/module/bcm2835_v4l2/parameters/debug <==
0

==> /sys/module/bcm2835_v4l2/parameters/max_video_height <==
2464

==> /sys/module/bcm2835_v4l2/parameters/max_video_width <==
3280

==> /sys/module/bcm2835_v4l2/parameters/video_nr <==
-1,-1
```
</details>

<details>
<summary>Full logs using verbose mode</summary>

```
2021-11-01 16:38:51,310 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/inc/logger.h:44
	level:DEBUG
2021-11-01 16:38:51,310 [NOTICE] - /home/pi/v4l2rtspserver/main.cpp:294
	Version: 1f0c768 live555 version:2021.11.01
2021-11-01 16:38:51,311 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:36
	Create V4L2 Source.../dev/video0
2021-11-01 16:38:51,312 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:bm2835 mmal capabilities:85200005 mandatory:4000001
2021-11-01 16:38:51,312 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video0 support capture
2021-11-01 16:38:51,312 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
	/dev/video0 support read/write
2021-11-01 16:38:51,312 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video0 support streaming
2021-11-01 16:38:51,313 [DEBUG ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:71
	/dev/video0:JPEG size:1640x1232 bufferSize:2050048
2021-11-01 16:38:51,318 [ERROR ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:207
	/dev/video0: Cannot set format:H264 Invalid argument
2021-11-01 16:38:51,319 [ERROR ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:207
	/dev/video0: Cannot set format:MJPG Invalid argument
2021-11-01 16:38:51,324 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
	/dev/video0:JPEG size:1640x1232 bufferSize:2050048
2021-11-01 16:38:51,325 [DEBUG ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:71
	/dev/video0:JPEG size:1640x1232 bufferSize:2050048
2021-11-01 16:38:51,325 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:246
	fps:1/25
2021-11-01 16:38:51,326 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:247
	nbBuffer:1
2021-11-01 16:38:51,326 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
2021-11-01 16:38:51,357 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
2021-11-01 16:38:51,357 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:0 size:2050048 offset:0
2021-11-01 16:38:51,358 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:1 size:2050048 offset:2052096
2021-11-01 16:38:51,358 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:2 size:2050048 offset:4104192
2021-11-01 16:38:51,359 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:3 size:2050048 offset:6156288
2021-11-01 16:38:51,360 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:4 size:2050048 offset:8208384
2021-11-01 16:38:51,360 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:5 size:2050048 offset:10260480
2021-11-01 16:38:51,361 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:6 size:2050048 offset:12312576
2021-11-01 16:38:51,361 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:7 size:2050048 offset:14364672
2021-11-01 16:38:51,362 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:8 size:2050048 offset:16416768
2021-11-01 16:38:51,363 [INFO  ] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:92
	Device /dev/video0 buffer idx:9 size:2050048 offset:18468864
VIDIOC_STREAMON: Operation not permitted
2021-11-01 16:38:51,810 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
	Device /dev/video0
```
</details>

I will keep going at trying to see which bit I missed, but it does seem to be very odd. Using a `modprobe -r bcm2835_v4l2` removing the `.conf` file and then a `modprobe bcm2835_v4l2` does bring the feed back, but I am still stuck at 640x480 resolution.

mystcb, 2021-11-01:
Woo! Success! Using the changes @zeretar recommended, I noticed that setting the max_video_height/width a high did not seem to work for the specific video settings I was trying to achieve.

Eventually, I set the options to exactly the resolution I was trying to achieve for the video (1640x1232) and it worked! 

```
$ sudo nano /etc/modprobe.d/bcm2835_v4l2.conf
options bcm2835_v4l2 max_video_height=1232
options bcm2835_v4l2 max_video_width=1640

$ sudo modprobe -r bcm2835_v4l2
$ sudo modprobe bcm2835_v4l2

$ tail -n +1 /sys/module/bcm2835_v4l2/parameters/*
==> /sys/module/bcm2835_v4l2/parameters/debug <==
0

==> /sys/module/bcm2835_v4l2/parameters/max_video_height <==
1232

==> /sys/module/bcm2835_v4l2/parameters/max_video_width <==
1640

==> /sys/module/bcm2835_v4l2/parameters/video_nr <==
-1,-1
```

azsde, 2021-11-03:
Amazing ! Thanks for the fix guys :)