mpromonet/v4l2rtspserver #198: make fails in ServerMediaSubsession.cpp.o: RawVideoRTPSink.hh:35:3: note:   candidate expects 8 arguments, 7 provided

bschaefer99, 2020-08-09:
```
[  5%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o
/root/v4l2rtspserver/src/ServerMediaSubsession.cpp: In static member function ‘static RTPSink* BaseServerMediaSubsession::createSink(UsageEnvironment&, Groupsock*, unsigned char, const string&, V4L2DeviceSource*)’:
/root/v4l2rtspserver/src/ServerMediaSubsession.cpp:87:146: error: no matching function for call to ‘RawVideoRTPSink::createNew(UsageEnvironment&, Groupsock*&, unsigned char&, int, int, int, const char*)’
   videoSink = RawVideoRTPSink::createNew(env, rtpGroupsock, rtpPayloadTypeIfDynamic, source->getHeight(), source->getWidth(), 8, sampling.c_str());
                                                                                                                                                  ^
In file included from /root/v4l2rtspserver/live/liveMedia/include/liveMedia.hh:99,
                 from /root/v4l2rtspserver/inc/ServerMediaSubsession.h:20,
                 from /root/v4l2rtspserver/src/ServerMediaSubsession.cpp:14:
/root/v4l2rtspserver/live/liveMedia/include/RawVideoRTPSink.hh:35:3: note: candidate: ‘static RawVideoRTPSink* RawVideoRTPSink::createNew(UsageEnvironment&, Groupsock*, u_int8_t, unsigned int, unsigned int, unsigned int, const char*, const char*)’
   createNew(UsageEnvironment& env, Groupsock* RTPgs, u_int8_t rtpPayloadFormat,
   ^~~~~~~~~
/root/v4l2rtspserver/live/liveMedia/include/RawVideoRTPSink.hh:35:3: note:   candidate expects 8 arguments, 7 provided
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:2715: CMakeFiles/libv4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

As far as i can tell, i have all the pre-requisites in place.

mpromonet, 2020-08-09:
Hi Brian,

Yes you are right, live555 change its interface removing the default last argument and changing the order.
I will fix this.

Best Regards,
Michel.

closed 2020-08-09 by mpromonet

bschaefer99, 2020-08-09:
Thanks for the quick response, @mpromonet ! I was just double checking my dependencies when i saw your update.

bschaefer99, 2020-08-09:
Got 'em!
`[100%] Built target v4l2rtspserver`

commit 820c8e46 (2020-08-09): fix #198: give the 8 paremeters to RawVideoRTPSink::createNew using the new ordering
commit de5a8467 (2020-08-09): fix #198: give the 8 paremeters to RawVideoRTPSink::createNew using the new ordering