mpromonet/v4l2rtspserver #199: v4l2rtspserver (Docker) on Raspi having a problem with very low fps

mrBrutus, 2020-08-25:
Hi

I'm running v4l2rtspserver as Docker container on an old Rasberry Pi B rev2.

As I am planning to use this as a 1080x1920 camera source for [Frigate](https://github.com/blakeblackshear/frigate) object detection running on a Raspberry Pi 4 (+ Coral USB Accelerator) I want to use a *very low fps* setting. But somehow I cannot get this to work.

If I set `-F15` the stream is looking fine in VLC. But when I set `-F2`  (2 fps) the stream is not working (hardly any image, lots of lost frames). Any idea what I am doing wrong?

`docker-compose.yaml`:
```yaml
version: '3'
services:
  v4l2rtspserver:
    restart: always
    image: mpromonet/v4l2rtspserver
    container_name: v4l2rtspserver
    ports:
      - "8554:8554"
    devices:
      - "/dev/video0:/dev/video0"
    command:
      - -F2       # -F15 is working fine
      - -W1920
      - -H1080
```

mpromonet, 2020-08-26:
Hi,

Maybe you should look to the log (with higher verbosity level) ? Maybe the RTSP client is expected for some frame rate ?

Best Regards,
Michel.

mrBrutus, 2020-08-26:
Hi Michel

Thanks for pointing me in the right direction. 
- After setting `:network-caching=5000` (ms) in VLC, the stream is shown properly. 

But sadly this setting causes a huge delay.

The logs in v4l2rtspserver did not show problems using verbose level `-v` but the log in VLC did show a lot of issues when using `:network-caching=1000` :

```
main error: buffer deadlock prevented
direct3d11 warning: failed to get the 0 MatchingDeviceId (2)
d3d11va warning: failed to get the 0 MatchingDeviceId (2)
avcodec info: Using D3D11VA (Intel(R) HD Graphics 5500, vendor 8086(Intel), device 1616, revision 9) for hardware decoding
main warning: picture is too late to be displayed (missing 2241 ms)
main warning: picture is too late to be displayed (missing 2250 ms)
main warning: picture is too late to be displayed (missing 2253 ms)
main warning: picture is too late to be displayed (missing 2260 ms)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
main warning: picture is too late to be displayed (missing 7232 ms)
main warning: picture is too late to be displayed (missing 7252 ms)
main warning: picture is too late to be displayed (missing 7233 ms)
main warning: picture is too late to be displayed (missing 7274 ms)
main warning: picture is too late to be displayed (missing 2277 ms)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
avcodec error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
```

Do you think there is a way to get the same level of delay for video at such low fps? I tried tweaking various VLC parameters but could not improve it.

mpromonet, 2020-08-27:
Hi,

Did you tried `--no-drop-late-frames --no-skip-frames` ?

Best Regards
Michel.

closed 2022-02-06 by mpromonet