mpromonet/v4l2rtspserver #201: various error compiling for RPI B+

edefranco, 2020-09-09:
Hi, I try to install rpos on a RPI B+ with wheezy, gcc updated to 4.9.2, cmake 3.0.2 and kernel 4.1.19+, npm 8.4.0 and have some compiling problems with v4l2rtspserver
1) /root/rpos/v4l2rtspserver/main.cpp: In function ‘std::string getV4l2Alsa(const string&)’:
/root/rpos/v4l2rtspserver/main.cpp:157:58: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
deviceid = std::string(std::istreambuf_iterator{ifsd}, {});

Resolved changing in CMakeList.txt the row:
set(CMAKE_CXX_STANDARD 11)

into

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")

2) -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Configuring done
CMake Error at CMakeLists.txt:25 (add_executable):
Target "v4l2rtspserver" links to target "Threads::Threads" but the target
was not found. Perhaps a find_package() call is missing for an IMPORTED
target, or an ALIAS target is missing?

Resolved installing pkg-config and (but I am not sure that this modify is really correct) changing in CMakeList.txt this:

find_package (Threads)
target_link_libraries (${PROJECT_NAME} Threads::Threads)

to

find_package (Threads)
target_link_libraries (${PROJECT_NAME})

if (TARGET Threads::Threads)
   target_link_libraries (${PROJECT_NAME} Threads::Threads)
endif()

3) NOT RESOLVED !
Linking CXX executable v4l2rtspserver
/usr/bin/ld: liblibv4l2rtspserver.a(DeviceSource.cpp.o): undefined reference to symbol 'pthread_join@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/v4l2rtspserver.dir/build.make:89: set di istruzioni per l'obiettivo "v4l2rtspserver" non riuscito
make[2]: *** [v4l2rtspserver] Errore 1
CMakeFiles/Makefile2:96: set di istruzioni per l'obiettivo "CMakeFiles/v4l2rtspserver.dir/all" non riuscito
make[1]: *** [CMakeFiles/v4l2rtspserver.dir/all] Errore 2
Makefile:147: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2

Looking here for similar problem:
microsoft/cpprestsdk#927

I think I understand that the problem maybe is correlated to the linking order of pthreads ... but I am no sure also about my previous modified to CMakeList.txt about Thread ...

Any help is greatly appreciated

Emilio

mpromonet, 2020-09-09:
Hi Emilio,

It is probably simpler to cross compile
About thread you may try `-pthread`

Best Regards,
Michel.

edefranco, 2020-09-09:
I Michel, thank for your answer.
No, I am compiling directly on RPI, no docker, no snap.
I download from git and give cmake . && make.
Where I can put -pthread as you suggested ?

Emilio

mpromonet, 2020-09-10:
Hi Emilio,

In CFLAGS & LDFLAGS this is what I guess makes the abstraction `Threads`.

Best Regards,
Michel.

edefranco, 2020-09-10:
Hi Michel,

I modified again the CMakeList.txt as:

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")

and the pthread section now is:

set(CMAKE_THREAD_PREFER_PTHREAD TRUE)
set(THREADS_PREFER_PTHREAD_FLAG TRUE)
find_package (Threads)

target_link_libraries (${PROJECT_NAME} ${CMAKE_THREAD_LIBS_INIT})

if (TARGET Threads::Threads)
   target_link_libraries (${PROJECT_NAME} Threads::Threads)
endif()

and (but I don't know if this is important) I have moved the ptheard after librtsp.

The compilation ending whitout error ! VERY GOOD!

I write this notes in the eventuality that some one encounter my same problem.

Thank again, know staring the testing (I hope all working fine)

Emilio

closed 2020-09-19 by mpromonet