mpromonet/v4l2rtspserver #203: Stop and start the stream at will

Dipendra77, 2020-09-28:
How do I stop and start an rtsp stream at will on raspberry Pi ?  I am able to stream using the command line script, but I have a use case where I want to start and stop the rtsp stream from my python script running on the pi when I want. How do I achieve that ?  
I am very new to this. ANy help or suggestion would be appreciated.

Thank you in advance.

edefranco, 2020-10-01:
A simple way can be call a bash script that get the v4l2rtspserver pid (ps -e | grep v4l2rtspserver | awk '{print $1}' and kill it ...

mpromonet, 2020-10-01:
Hi,

There is a systemd script to start as a deamon. So you can use systemctl start v4l2rtspserver to start and systemctl stop v4l2rtspserver to stop it.

Using snap distribution allow to control start & stop using snap start/stop v4l2-rtspserver

Best Regards
Michel