mpromonet/v4l2rtspserver #206: Intermittent viewing issue on Raspberry Pi 3B+

mpv1953, 2020-10-05:
Hello:
Yes, the title is vague because I'm not what the problem is.  I installed the program on an RPi 3B+ running Buster.  It works, but every few seconds (ranges from 15 seconds to about two minutes) my monitor screen goes blank.
Sometimes, the "normal" level logging shows:
`root@pisc04:/tmp#
root@pisc04:/tmp# tail -500f v4l2rtsp_201005_1300.log
log level:500`
And others, it's this:
`log level:500
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:5771 bufferSize:205744
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 5771 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:5473 bufferSize:205446
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 5473 bytes!
AddH26xMarkerFilter::afterGettingFrame(): buffer too small truncated:4115 bufferSize:204088
MPEG2TransportStreamFromESSource: input buffer too small; increase "MPEG2TransportStreamFromESSource::maxInputESFrameSize" by at least 4115 bytes!`

The build was successful.  I run the application with this script:
`root@pisc04:/usr/local/bin# cat runsc.sh
#!/bin/bash
#

LFSuffix=`date +"%y%m%d_%H%M"`
LF=/tmp/v4l2rtsp_${LFSuffix}.log
MP=/sbin/modprobe
NH=/usr/bin/nohup
PG=/usr/bin/pgrep
PROGNAME=v4l2rtspserver
SU=/usr/bin/sudo
SV='/usr/local/bin/v4l2rtspserver'
# VB is for verbosity
VB=''                           # None
# VB='-v'                       # Kinda
# VB='-vv'                      # REALLY

set -xv

if [ -z "$( pgrep -f ${PROGNAME} )" ]; then
  if [ "$1" = "start" ]; then
    echo "START   - Start ${PROGNAME} in Background"
    ${SU} ${MP} -v bcm2835-v4l2
#   ${SU} ${NH} ${SV} -U pi:Ourhou5e -p 8888 -S -F 25 -W 1280 -H 720 /dev/video0 ${VB} > ${LF} 2>&1 &
#   ${SU} ${NH} ${SV} -U pi:Ourhou5e -p 8888 -F 25 -W 1280 -H 720 /dev/video0 ${VB} > ${LF} 2>&1 &
#   ${SU} ${NH} ${SV} -U pi:Ourhou5e -p 8888 -s -W 1280 -H 720 -F 20 /dev/video0 ${VB} > ${LF} 2>&1 &
#   ${SU} ${NH} ${SV} -U pi:Ourhou5e -p 8888 -S -W 1280 -H 720 -F 20 /dev/video0 ${VB} > ${LF} 2>&1 &
    ${SU} ${NH} ${SV} -U pi:Ourhou5e -p 8888 -W 1280 -H 720 -F 20 /dev/video0 ${VB} > ${LF} 2>&1 &
  fi
else
  if [ "$1" = "stop" ]; then
    echo "STATUS  - Stopping ${PROGNAME} ...."
    progPID=$( ${PG} -f ${PROGNAME} )
    ${SU} kill -9 $progPID
  fi
fi
#
if [ -z "$( pgrep -f ${PROGNAME} )" ]; then
    echo "STATUS  - ${PROGNAME} is Not Running ..."
    echo "INFO    - To Start ${PROGNAME} execute command below"
    echo "INFO    - $0 start"
  else
    progPID=$(pgrep -f ${PROGNAME})
    echo "STATUS  - ${PROGNAME} is Running ..."
    echo "STATUS  - PID is $progPID"
    echo "INFO    - To Stop ${PROGNAME} execute command below"
    echo "INFO    - $0 stop"
fi

set +xv

#
#       ###EOF###
root@pisc04:/usr/local/bin#`

I use this camera for exterior home monitoring, and it's aggravating to have the screen go blank, which is usually when there's something interesting.
You can see from the script that I've been trying different options, but none seem to work.

Do you have any suggestions/recommendations?
Thanks,
MIke

mpv1953, 2020-10-05:
Sorry about the formatting.  I don't know why it did that.
More Info:
`root@pisc04:/usr/local/bin#
root@pisc04:/usr/local/bin# free -h
              total        used        free      shared  buff/cache   available
Mem:          874Mi       111Mi       514Mi       6.0Mi       247Mi       692Mi
Swap:          99Mi          0B        99Mi
root@pisc04:/usr/local/bin#
root@pisc04:/usr/local/bin# swapon
NAME      TYPE SIZE USED PRIO
/var/swap file 100M   0B   -2
root@pisc04:/usr/local/bin#
root@pisc04:/usr/local/bin# uname -a
Linux pisc04 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
root@pisc04:/usr/local/bin#
root@pisc04:/usr/local/bin# /usr/local/bin/v4l2rtspserver -V
0.2.0-27-gb7a34ff
root@pisc04:/usr/local/bin#
root@pisc04:/boot# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000faad28f5
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
root@pisc04:/boot#
`

biodrone, 2022-05-24:
Having the same issue here, did you find a fix @mpv1953 ?

mpv1953, 2022-05-25:
Hi @biodrone:
No I did not.  I switched gears and went with local recording instead of using a web browser.
Cheers,
MIke

mpromonet, 2022-05-25:
Hi,

[MPEG2TransportStreamFromESSource::maxInputESFrameSize](https://github.com/rgaufman/live555/blob/master/liveMedia/MPEG2TransportStreamFromESSource.cpp#L76) is set to 100000.
If you have bigger image, you can try to increase it.

Best Regards,
Michel.

biodrone, 2022-05-25:
Hi Michael,

Thanks for the follow up! I'm using the docker container, is there a flag to pass to change this parameter or will I have to compile it myself?

Thanks,
Josh