mpromonet/v4l2rtspserver #207: Memory leak in HLS server

tachang, 2020-10-11:
I see things growing slowly eating more memory until it cannot allocate anymore memory.
Is this a known issue?

```
# /usr/bin/v4l2rtspserver -S2 /dev/video3
log level:500

[NOTICE] main.cpp:385
	Version: 1 live555 version:2016.01.29
[NOTICE] main.cpp:426
	Create V4L2 Source.../dev/video3
[NOTICE] src/V4l2Device.cpp:133
	driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:135
	/dev/video3 support output
[NOTICE] src/V4l2Device.cpp:136
	/dev/video3 support capture
[NOTICE] src/V4l2Device.cpp:138
	/dev/video3 support read/write
[NOTICE] src/V4l2Device.cpp:139
	/dev/video3 support streaming
[NOTICE] src/V4l2Device.cpp:225
	/dev/video3:H264 size:1920x1080 bufferSize:8294400
[NOTICE] src/V4l2Device.cpp:246
	fps:1/25
[NOTICE] src/V4l2Device.cpp:247
	nbBuffer:2
[NOTICE] src/V4l2MmapDevice.cpp:49
	Device /dev/video3
[NOTICE] src/V4l2MmapDevice.cpp:73
	Device /dev/video3 nb buffer:2
[NOTICE] main.cpp:449
	Create Source .../dev/video3
[NOTICE] inc/V4l2RTSPServer.h:82
	Play this stream using the URL "rtsp://192.168.0.179:8554/ts"
[NOTICE] main.cpp:527
	HLS       http://192.168.0.179:8554/ts.m3u8
[NOTICE] main.cpp:528
	MPEG-DASH http://192.168.0.179:8554/ts.mpd
[NOTICE] inc/V4l2RTSPServer.h:82
	Play this stream using the URL "rtsp://192.168.0.179:8554/unicast"
[NOTICE] src/DeviceSource.cpp:93
	send M3u8 playlist:ts
send M3u8 playlist:ts
begin threadseek seekNPT:0 slice:0 numBytes:359644
seek seekNPT:2 slice:1 numBytes:593516
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:4 slice:2 numBytes:593516
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:6 slice:3 numBytes:592388
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:6 slice:3 numBytes:592388
seek seekNPT:8 slice:4 numBytes:593892
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:8 slice:4 numBytes:593892
seek seekNPT:10 slice:5 numBytes:594456
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:10 slice:5 numBytes:594456
seek seekNPT:12 slice:6 numBytes:596712
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:12 slice:6 numBytes:596712
seek seekNPT:14 slice:7 numBytes:594456
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:14 slice:7 numBytes:594456
seek seekNPT:16 slice:8 numBytes:594456
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:16 slice:8 numBytes:594456
seek seekNPT:18 slice:9 numBytes:594644
send M3u8 playlist:ts
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:18 slice:9 numBytes:594644
seek seekNPT:20 slice:10 numBytes:596900
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:20 slice:10 numBytes:596900
send M3u8 playlist:ts
seek seekNPT:22 slice:11 numBytes:595396
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:22 slice:11 numBytes:595396
seek seekNPT:24 slice:12 numBytes:595584
send M3u8 playlist:ts
seek seekNPT:24 slice:12 numBytes:595584
seek seekNPT:26 slice:13 numBytes:595020
send M3u8 playlist:ts
seek seekNPT:26 slice:13 numBytes:595020
seek seekNPT:28 slice:14 numBytes:596336
send M3u8 playlist:ts
seek seekNPT:28 slice:14 numBytes:596336
seek seekNPT:30 slice:15 numBytes:597840
send M3u8 playlist:ts
seek seekNPT:28 slice:14 numBytes:596336
seek seekNPT:30 slice:15 numBytes:597840
seek seekNPT:32 slice:16 numBytes:596336
seek seekNPT:34 slice:17 numBytes:594832
send M3u8 playlist:ts
seek seekNPT:34 slice:17 numBytes:594832
seek seekNPT:36 slice:18 numBytes:592764
send M3u8 playlist:ts
seek seekNPT:34 slice:17 numBytes:594832
seek seekNPT:36 slice:18 numBytes:592764
seek seekNPT:38 slice:19 numBytes:598404
seek seekNPT:40 slice:20 numBytes:595020
send M3u8 playlist:ts
send M3u8 playlist:ts
seek seekNPT:40 slice:20 numBytes:595020
seek seekNPT:42 slice:21 numBytes:595960
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted
```

mpromonet, 2020-10-11:
Hi tachang,

This was introduced adapting to remove in live555 of TCPStreamSink, in commit https://github.com/mpromonet/v4l2rtspserver/commit/be67798772d00929c4af6b0b31f2bbea6d49c3b4.
I will look at that.

Best Regards,
Michel.

closed 2020-10-11 by mpromonet

tachang, 2020-10-11:
Thank you so much! This commit definitely solved the issue and memory is no longer growing unbounded.

commit 16c21de9 (2020-10-11): fix #207: fix memory leak streaming HLS