mpromonet/v4l2rtspserver #210: Information on implementing compressed audio codecs

tachang, 2020-10-17:
Hi,
  Thought I'd put this question out there in case other people might be interested in implementing something like this.
Basically I am trying to get Opus audio compression working. The Dafang hacks v4l2rtspserver fork managed this but I am having trouble replicating it. My question is over here:

https://stackoverflow.com/questions/64406476/having-a-hard-time-getting-opus-to-stream-properly

And I have links to some more code I'm working on here:
https://github.com/tachang/v4l2rtspserver/pull/1/files

@mpromonet Just putting this out for your input since I might try to submit a PR to get this mainlined. I know the approach is definitely hacky so I'd rework that but curious to know if you have any ideas.

mpromonet, 2020-10-18:
Hi @tachang 

Supporting opus may be interesting, but your modification may need improvement (detecting if opus is available in CMakefile, moving opus stuff in a inherited class of ALSA).
I saw Dafang Hack fork https://github.com/Dafang-Hacks/Main/tree/master/v4l2rtspserver-master make encoding stuff.

Anyway PR are welcome, it could interest other people.

Best Regards,
Michel.

tachang, 2020-10-18:
@mpromonet Understood. One question I had is what is the contract for the ::read() method?

I am having a lot of trouble getting this Opus code to work. Am I correct that the read() method is called 25 times per second? Or basically the amount of time set as the streaming frames per second? Or is it something else? I am having trouble finding the code that uses this ALSA C++ class and calls it.

tachang, 2020-10-28:
@mpromonet  Does it make more sense to implement it as a change to ALSACapture? Perhaps with options to enable compression? I think this makes more sense than inheritance. I went down the route of inheritance where the new class MP3ALSACapture extends from ALSACapture but I don't think this is a good way to model it.

closed 2021-04-22 by tachang