mpromonet/v4l2rtspserver #211: libasound2-dev dependency in the deb package

villytiger, 2020-10-24:
Thank you for the great project!  I have a couple of questions regarding alsa support.

Why does the deb package depends on libasound2-dev? I think that a target system should contain libasound2 only.

Is there any particular reason why v4l2rtspserver-0.2.0-Linux-armv7l.deb is built without alsa support? Would you mind adding alsa support to it?

mpromonet, 2020-10-24:
Hi Ilya,

The dependency of deb to libasound2-dev is an error, as you note, it should be libasound2, something to do.

In the past, the deb files for arm were built with alsa support, this need to cross-compile ALSA, I removed this due to difficulty to maintain cross-compile chain, it may be nice to retry that....

There is 2 ways to have alsa support, one is to build the executable,  the other is to use snap deliverable.

Best Regards,
Michel.

villytiger, 2020-10-25:
Hi Michel,

I've just realized that the arm build is targeting raspbian on armv6. At first, I thought that we could use alsa files from raspbian packages. But the cross toolchain for raspbian is so old that the raspbian mirror doesn't contain proper packages.

What do you think about building one more package targeting armv7? In this case alsa files can be used from regular debian packages with the help of multiarch support. I can build v4l2rtspserver on Debian Stretch and Buster (and therefore targeting them) with alsa support using the following script:
```
dpkg --add-architecture armhf
apt update
apt install -y --no-install-recommends cmake autoconf automake libtool git make g++-arm-linux-gnueabihf pkg-config
apt install -y --no-install-recommends ca-certificates libasound2-dev:armhf
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver
# Needed on Debian Stretch,  no need on Debian Buster
echo 'set(THREADS_PTHREAD_ARG "0" CACHE STRING "Result from TRY_RUN" FORCE)' >>raspberry.toolchain
cmake -DCMAKE_TOOLCHAIN_FILE=raspberry.toolchain .
make
cpack .
```
The same can be done on Ubuntu but in a more hacky way. It is also possible to add alsa support to mips package with the same technique.

I can create a pull request if you want.

mpromonet, 2020-10-25:
Hi Ilya,

The deb in release is for armv7, but, as you noticed used a very old toolchain. Last time I tried using the ubuntu g++-arm-linux-gnueabihf crosscompiler build executable with higher glibc/stdc++ than the one running on a raspberry B. In another project I am using toolchain from https://sourceforge.net/projects/raspberry-pi-cross-compilers.

Best Regards,
Michel.

closed 2020-12-09 by mpromonet