mpromonet/v4l2rtspserver #212: v4l2rtspserver crash?

s-drahcir, 2020-10-24:
Hello :)

Apologies if this *isn't* your issue, but as I only really run v4l2rtspserver, I assumed it was related. I'm also very new to this...

I've recently updated v4l2rtspserver and the OS (I  use a lite version to help with available resources) of my Raspberry Pi W. I've had it running for a day or so and found the stream had stopped. Upon doing dmesg:

[179114.279715] 8<--- cut here ---
[179114.281298] Unable to handle kernel paging request at virtual address bf95aed0
[179114.282878] pgd = f43dc0ef
[179114.284445] [bf95aed0] *pgd=00000000
[179114.286097] Internal error: Oops: 80000005 [#1] ARM
[179114.288719] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 raspberrypi_hwmon cfg80211 rfkill bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) snd_pcm snd_timer videodev snd mc vc_sm_cma(C) uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[179114.300802] CPU: 0 PID: 705 Comm: v4l2rtspserver Tainted: G         C        5.4.72+ #1354
[179114.307660] Hardware name: BCM2835
[179114.310202] PC is at 0xbf95aed0
[179114.313026] LR is at video_usercopy+0x498/0x5e0 [videodev]
[179114.315942] pc : [<bf95aed0>]    lr : [<bf11eec4>]    psr: 60000013
[179114.318872] sp : cfef9df0  ip : cfef9df0  fp : cfef9ebc
[179114.321826] r10: cfef9e0c  r9 : 00000000  r8 : 00000003
[179114.324657] r7 : 00000000  r6 : 00000044  r5 : 00000000  r4 : c0445611
[179114.327570] r3 : 00000000  r2 : bf1406a0  r1 : 00000000  r0 : d142c058
[179114.330399] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[179114.333107] Control: 00c5387d  Table: 0c644008  DAC: 00000055
[179114.335381] Process v4l2rtspserver (pid: 705, stack limit = 0xcdc60f5f)
[179114.337643] Stack: (0xcfef9df0 to 0xcfefa000)
[179114.339899] 9de0:                                     00000000 00000000 cc5446e0 b57acc28
[179114.344517] 9e00: 00000000 00000000 c0b432c8 00000005 00000001 00012db3 00002001 00000001
[179114.349085] 9e20: 0002bbaa 0002c66f 00000000 00000000 00000000 00000000 00284673 00000001
[179114.353701] 9e40: 009f6000 001fe000 00000000 00000000 c0186288 c018e2cc 00000000 c01c47e4
[179114.359391] 9e60: 00000000 cfef9ea8 cfef9e9c cfef9e78 c01c47e4 00000000 d0936360 c018e5b8
[179114.363991] 9e80: 00000000 d0936360 0e17f34f c0aa7028 cfef9f0c bf11f00c d17e90c0 cc5446e0
[179114.368810] 9ea0: 00000004 cc5446e0 00000004 00000000 cfef9ecc cfef9ec0 bf11f028 bf11ea38
[179114.376312] 9ec0: cfef9ee4 cfef9ed0 bf117148 bf11f018 b57acc28 d17e90c0 cfef9f6c cfef9ee8
[179114.382517] 9ee0: c01e3828 bf117108 cfef9fb0 b4c01000 00000805 d245d580 cd4c2700 c07b31fc
[179114.389575] 9f00: cfef9f5c cfef9f10 c07b31fc c0063238 00000000 cfef9f20 cfef9f24 b57acd48
[179114.396748] 9f20: 00000000 c01ee5ac 00067100 01e9e9f8 c0445611 b57acc28 c00091a4 c0aa7028
[179114.402986] 9f40: cfef9f5c cc5446e1 00000000 c0445611 b57acc28 cc5446e0 00000004 00000000
[179114.408897] 9f60: cfef9f94 cfef9f70 c01e3cb8 c01e34b0 00067100 01e9e9f8 b6fa5950 00000036
[179114.414841] 9f80: c00091a4 cfef8000 cfef9fa4 cfef9f98 c01e3cf4 c01e3c58 00000000 cfef9fa8
[179114.421064] 9fa0: c0009000 c01e3ce8 00067100 01e9e9f8 00000004 c0445611 b57acc28 00000004
[179114.427897] 9fc0: 00067100 01e9e9f8 b6fa5950 00000036 beea2692 b57ad450 00000000 b57acc84
[179114.433415] 9fe0: 00111144 b57acc14 0006a56c b6c8d51c 40000010 00000004 00000000 00000000
[179114.441699] Backtrace:
[179114.445332] [<bf11ea2c>] (video_usercopy [videodev]) from [<bf11f028>] (video_ioctl2+0x1c/0x24 [videodev])
[179114.452418]  r10:00000000 r9:00000004 r8:cc5446e0 r7:00000004 r6:cc5446e0 r5:d17e90c0
[179114.459579]  r4:bf11f00c
[179114.463469] [<bf11f00c>] (video_ioctl2 [videodev]) from [<bf117148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[179114.469664] [<bf1170fc>] (v4l2_ioctl [videodev]) from [<c01e3828>] (do_vfs_ioctl+0x384/0x7a8)
[179114.475283]  r5:d17e90c0 r4:b57acc28
[179114.478012] [<c01e34a4>] (do_vfs_ioctl) from [<c01e3cb8>] (ksys_ioctl+0x6c/0x90)
[179114.483344]  r10:00000000 r9:00000004 r8:cc5446e0 r7:b57acc28 r6:c0445611 r5:00000000
[179114.489852]  r4:cc5446e1
[179114.492740] [<c01e3c4c>] (ksys_ioctl) from [<c01e3cf4>] (sys_ioctl+0x18/0x1c)
[179114.495357]  r9:cfef8000 r8:c00091a4 r7:00000036 r6:b6fa5950 r5:01e9e9f8 r4:00067100
[179114.500074] [<c01e3cdc>] (sys_ioctl) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[179114.507562] Exception stack(0xcfef9fa8 to 0xcfef9ff0)
[179114.510499] 9fa0:                   00067100 01e9e9f8 00000004 c0445611 b57acc28 00000004
[179114.516641] 9fc0: 00067100 01e9e9f8 b6fa5950 00000036 beea2692 b57ad450 00000000 b57acc84
[179114.523331] 9fe0: 00111144 b57acc14 0006a56c b6c8d51c
[179114.526836] Code: bad PC value
[179114.530128] ---[ end trace 7c7c6e9419b1f585 ]---

Doing top, it shows:

top - 18:27:40 up 2 days, 21:31,  1 user,  load average: 0.10, 0.11, 0.09
Tasks:  77 total,   1 running,  76 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.7 us,  1.0 sy,  0.0 ni, 97.3 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    368.4 total,    146.9 free,     51.8 used,    169.6 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    249.0 avail Mem

PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
 704 root      20   0   41284  24376  24112 S   1.3   6.5 715:51.33 v4l2rtsps+

uname -a shows:

Linux raspberrypi 5.4.72+ #1354 Mon Oct 19 13:34:33 BST 2020 armv6l GNU/Linux

Upon trying to restart it (I have it set up as a service), in dmesg, I see:

tachang, 2020-10-29:
This almost sounds like bad ram. Kernel isn't able to handle a paging request? Code: bad PC value means something is weird in the registers? Sounds odd. Can you run any memory tests against a rasp pi? Or even the sdcard if it has one.

mpromonet, 2020-11-09:
Hi,

This seems a crash in the v4l2 driver, isn't it ?

Best Regards,
Michel.

closed 2021-07-04 by mpromonet