mpromonet/v4l2rtspserver #215: RTCP？

gh2517956473, 2020-11-09:
Can this project use RTCP sessions?

 I have a need to synchronize video and gyroscope data

mpromonet, 2020-11-09:
Hi,

Yes it is based on live555 that manage RTCP.

Best Regards,
Michel.

gh2517956473, 2020-11-10:
So this means that I can receive the **NTP time** of RTCP sessions on the  **client**？
And RTCP sessions can be started by running the project in the following ways？

        modprobe v4l2loopback video_nr=10
	v4l2compress_h264 /dev/video0 /dev/video10 &
	v4l2rtspserver /dev/video10 &

We appreciate your help and your patience!

mpromonet, 2020-11-10:
Hi,

In the RTCP SR (sender report) you should get link between RTP time and NTP time.

Best Regards,
Michel.

closed 2020-11-29 by mpromonet