mpromonet/v4l2rtspserver #216: Can't build on raspberry pi 3

dirtdigger, 2020-11-26:
Hello, I'm trying to compile on a freshly updated rpi 3 but I'm running into problems. Here are the commands I've run:
```
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver
cmake .
```
So far so good, checks out v4l2wrapper and a couple of other things. I get a message saying that build files have been written. So I run

```
make
```

and then I get a bunch of errors:

> [  4%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/BasicHashTable.cpp.o
[  4%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/BasicTaskScheduler.cpp.o
[  5%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/BasicTaskScheduler0.cpp.o
[  5%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/BasicUsageEnvironment.cpp.o
[  6%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/BasicUsageEnvironment0.cpp.o
[  6%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/BasicUsageEnvironment/DelayQueue.cpp.o
[  7%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/UsageEnvironment/HashTable.cpp.o
[  7%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/UsageEnvironment/UsageEnvironment.cpp.o
[  8%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/UsageEnvironment/strDup.cpp.o
[  8%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/GroupEId.cpp.o
[  9%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/Groupsock.cpp.o
[  9%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/GroupsockHelper.cpp.o
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp: In function ‘int readSocket(UsageEnvironment&, int, unsigned char*, unsigned int, sockaddr_storage&)’:
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:370:45: error: ‘struct in6_addr’ has no member named ‘__u6_addr’; did you mean ‘in6_addr’?
    ((sockaddr_in6*)&fromAddress)->sin6_addr.__u6_addr.__u6_addr32[0] = 0;
                                             ^~~~~~~~~
                                             in6_addr
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:371:45: error: ‘struct in6_addr’ has no member named ‘__u6_addr’; did you mean ‘in6_addr’?
    ((sockaddr_in6*)&fromAddress)->sin6_addr.__u6_addr.__u6_addr32[1] = 0;
                                             ^~~~~~~~~
                                             in6_addr
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:372:45: error: ‘struct in6_addr’ has no member named ‘__u6_addr’; did you mean ‘in6_addr’?
    ((sockaddr_in6*)&fromAddress)->sin6_addr.__u6_addr.__u6_addr32[2] = 0;
                                             ^~~~~~~~~
                                             in6_addr
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:373:45: error: ‘struct in6_addr’ has no member named ‘__u6_addr’; did you mean ‘in6_addr’?
    ((sockaddr_in6*)&fromAddress)->sin6_addr.__u6_addr.__u6_addr32[3] = 0;
                                             ^~~~~~~~~
                                             in6_addr
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:206: CMakeFiles/libv4l2rtspserver.dir/live/groupsock/GroupsockHelper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
>

Any advice? Thanks!

mramirezd, 2020-11-27:
I have the same issue on a raspberry pi 4B 8GB

mpromonet, 2020-11-27:
Hi,

Looking to http://www.live555.com/liveMedia/public/changelog.txt it seems live555 is implementing ipv6. This may have some impact in the code.

You may try using an older release that you can find for instance https://download.videolan.org/pub/contrib/live555/ running cmake with -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/....

Best Regards,
Michel.

mramirezd, 2020-11-27:
Thanks for the quick response Michel! 

I cloned the v0.2.1 and ran the cmake with 
`cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2020.08.19.tar.gz`

Working perfectly now!

closed 2020-11-27 by mpromonet

dirtdigger, 2020-11-27:
Awesome, thank you! I figured it was probably a live555 issue, but I couldn't find an older version on their website.

commit f0f04b73 (2020-11-27): fix #216: allow to build with live555 newer than 2020.11.27 (2020.11.25 & 2020.11.26 doesnot build)