mpromonet/v4l2rtspserver #217: 1080x720 image to 640x480 image

gh2517956473, 2020-11-28:
V4L2 options：
	 -W width : V4L2 capture width (default 640)
	 -H height: V4L2 capture height (default 480)

Why 640x480 image from Camera is  the   **center  crop**  of the 1080x720 image，The camera's **field of view** has **narrowed**.

How can I  get 640x480 image Camera is  the   **resized  image** of the 1080x720 image？I want to keep  640x480 camera's field of view  **same as**  1080x720  camera.
Thank you very much！

mpromonet, 2020-11-28:
Hi gh2517956473,

v4l2rtspserver doesnot do any image modification (neither resize, crop, ...), this is done by your v4l2 driver.
A minimal ioctl is called to the V4L2 driver for format, width, height, fps. You can also configure the V4L2 format before running v4l2rtpserver and run it with `-f` this will keep the capture as it is.

Best Regards,
Michel.

gh2517956473, 2020-11-28:
Hi Michel
Because now I just **change **v4l2rtspserve options** -W -H**, the  field of view of  image ** is narrowed**.
If I configure the V4L2 format before running v4l2rtpserver and run it with -f,  can I make 640x480 image have **same field of view** as 1080x720 image? 
Thank you very much！

closed 2021-07-04 by mpromonet