mpromonet/v4l2rtspserver #220: Build failure: no matching function for call to ‘Groupsock::Groupsock(UsageEnvironment&, in_addr&, Port&, int&)

nettings, 2020-12-12:
Looks like a function signature changed in the latest live555 version? Here's the full build output:

[Groupsock build error.txt](https://github.com/mpromonet/v4l2rtspserver/files/5683385/Groupsock.build.error.txt)

nettings, 2020-12-12:
Had a hard time figuring out which version of live555 was used for this build. Checking on their website, I find the following files and timestamps:
``` 
  | live.2020.12.11.tar.gz | 2020-12-10 16:44 | 663K |  
  | live555-latest-md5.txt | 2020-12-10 16:44 | 33 |  
  | live555-latest.tar.gz | 2020-12-10 16:44 | 663K |
```

nettings, 2020-12-12:
not really related, but apparently this is not the only recent breakage due to live555 changes: #218 (already fixed by @mpromonet)

nettings, 2020-12-12:
Ok, found this in their FAQ: 'To see which version of the code you currently have, look at the file "liveMedia/include/liveMedia_version.hh".' One may be excused for overlooking that one :o)
My version is
#define LIVEMEDIA_LIBRARY_VERSION_STRING	"2020.12.11"
#define LIVEMEDIA_LIBRARY_VERSION_INT		1607644800

mpromonet, 2020-12-12:
Hi nettings,

live555 seems to change often of interface this days.... it doesnot even build with 2020.12.12.
You can build it with :
```
cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2020.12.11.tar.gz .
```

Best Regards,
Michel.

closed 2020-12-12 by mpromonet

nettings, 2020-12-12:
Dear Michel, sorry to hear that - it must be a lot of extra work for you. Thank you for the workaround, that solves my problem! Maybe you should update your README.md to prevent other users from hitting this issue. All best, Jörn

nettings, 2020-12-12:
OMG. They broke it again... Just posting this for your information, the workaround with the old live555 source works fine for me.

```[ 10%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/NetAddress.cpp.o
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp: In function ‘void copyAddress(sockaddr_storage&, const NetAddress*)’:
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp:99:8: error: ‘struct sockaddr_storage’ has no member named ‘ss_len’
     to.ss_len = sizeof (struct sockaddr_in);
        ^~~~~~
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp:103:8: error: ‘struct sockaddr_storage’ has no member named ‘ss_len’
     to.ss_len = sizeof (struct sockaddr_in6);
        ^~~~~~
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:232: CMakeFiles/libv4l2rtspserver.dir/live/groupsock/NetAddress.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
[  4%] Built target v4l2wrapper
[  5%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/NetAddress.cpp.o
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp: In function ‘void copyAddress(sockaddr_storage&, const NetAddress*)’:
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp:99:8: error: ‘struct sockaddr_storage’ has no member named ‘ss_len’
     to.ss_len = sizeof (struct sockaddr_in);
        ^~~~~~
/medianet/custom_builds/v4l2rtspserver/live/groupsock/NetAddress.cpp:103:8: error: ‘struct sockaddr_storage’ has no member named ‘ss_len’
     to.ss_len = sizeof (struct sockaddr_in6);
        ^~~~~~
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:232: CMakeFiles/libv4l2rtspserver.dir/live/groupsock/NetAddress.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

mpromonet, 2020-12-13:
Hi nettings,

This is quite uncommon, usually live555 doesnot evolve much, it seems there is work in progress about IPV6.
The approach of this project is to use latest version, often small refactoring vs big refactoring.
Anyway you suggestion seems meaningful, I will add a note in the README to build with an oldest live555 release (even this release comes from videolan)

Best Regards,
Michel.

commit ac37fc39 (2020-12-12): fix #220: allow to build with live555 2020.12.11