mpromonet/v4l2rtspserver #223: v4l2rtspserver on Jetson Nano

lagurus, 2021-01-15:
Hello,
just info for users which have problems with using v4l2rtspserver on Jetson Nano with IMX219 camera.

I have tried <i>v4l2loopback</i> and <i>v4l2tools</i> and recommended path:
<code>/dev/video0 (camera device)-> v4l2compress_h264 -> /dev/video10 (v4l2loopback device) -> v4l2rtspserver</code> without success.

But I have found solution using gstreamer:

`gst-launch-1.0 -v nvarguscamerasrc ! 'video/x-raw(memory:NVMM), format=NV12, width=640, height=480, framerate=30/1' ! nvv4l2h264enc bitrate=100000 ! h264parse config-interval=-1 ! 'video/x-h264,stream-format=(string)byte-stream' ! v4l2sink device=/dev/video3`

Then use `./v4l2rtspserver -v -F30 /dev/video3`

where /dev/video3 is loopback device created with _v4l2loopback_ (`modprobe v4l2loopback video_nr=3 maxbuffers=10`)

mpromonet, 2021-01-15:
Hi lagurus,

It is not really clear what means without success, does is concern v4l2rtspserver ? or v4l2compress ?
Maybe the logs could help you ?

Best Regards,
Michel.

ksaye, 2021-01-19:
How are you compiling v4l2rtspserver on the Jetson Nano?  I get warnings with liveMedia and then a failure with v4l2rtspserver.  I assume it is not ready for ARM 64.

```
uname -a
Linux jetson 4.9.140-tegra #1 SMP PREEMPT Tue Oct 27 21:02:37 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
```

**Warnings:**
```
[ 26%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H263plusVideoStreamParser.cpp.o
/home/ksaye/v4l2rtspserver/live/liveMedia/H263plusVideoStreamParser.cpp: In member function ‘int H263plusVideoStreamParser::parseH263Frame()’:
/home/ksaye/v4l2rtspserver/live/liveMedia/H263plusVideoStreamParser.cpp:201:66: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 4 has type ‘long int’ [-Wformat=]
          "h263reader:", bufferEnd - fTo + ADDITIONAL_BYTES_NEEDED);
                                                                  ^
[ 27%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoFileServerMediaSubsession.cpp.o
[ 27%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoFileSink.cpp.o
[ 28%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoRTPSink.cpp.o
[ 28%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoRTPSource.cpp.o
[ 29%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoStreamDiscreteFramer.cpp.o
[ 29%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264VideoStreamFramer.cpp.o
[ 30%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264or5VideoFileSink.cpp.o
[ 30%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264or5VideoRTPSink.cpp.o
[ 31%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264or5VideoStreamDiscreteFramer.cpp.o
[ 31%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H264or5VideoStreamFramer.cpp.o
[ 31%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoFileServerMediaSubsession.cpp.o
[ 32%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoFileSink.cpp.o
[ 32%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoRTPSink.cpp.o
[ 33%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoRTPSource.cpp.o
[ 33%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoStreamDiscreteFramer.cpp.o
[ 34%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/H265VideoStreamFramer.cpp.o
[ 34%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/HLSSegmenter.cpp.o
[ 35%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/HMAC_SHA1.cpp.o
[ 35%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/InputFile.cpp.o
[ 36%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/JPEG2000VideoRTPSink.cpp.o
[ 36%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/JPEG2000VideoRTPSource.cpp.o
[ 37%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/JPEGVideoRTPSink.cpp.o
[ 37%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/JPEGVideoRTPSource.cpp.o
[ 37%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/JPEGVideoSource.cpp.o
[ 38%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/Locale.cpp.o
[ 38%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MIKEY.cpp.o
[ 39%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADU.cpp.o
[ 39%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADURTPSink.cpp.o
[ 40%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADURTPSource.cpp.o
[ 40%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADUTranscoder.cpp.o
[ 41%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADUdescriptor.cpp.o
[ 41%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3ADUinterleaving.cpp.o
[ 42%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3AudioFileServerMediaSubsession.cpp.o
[ 42%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3AudioMatroskaFileServerMediaSubsession.cpp.o
[ 42%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3FileSource.cpp.o
[ 43%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3Internals.cpp.o
[ 43%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3InternalsHuffman.cpp.o
[ 44%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3InternalsHuffmanTable.cpp.o
[ 44%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3StreamState.cpp.o
[ 45%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MP3Transcoder.cpp.o
[ 45%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MPEG1or2AudioRTPSink.cpp.o
[ 46%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MPEG1or2AudioRTPSource.cpp.o
[ 46%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MPEG1or2AudioStreamFramer.cpp.o
[ 47%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MPEG1or2Demux.cpp.o
[ 47%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/liveMedia/MPEG1or2DemuxedElementaryStream.cpp.o
[... cut ...]
/home/ksaye/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:91:14: note:   candidate expects 3 arguments, 2 provided
CMakeFiles/libv4l2rtspserver.dir/build.make:4862: recipe for target 'CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o' failed
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/libv4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/libv4l2rtspserver.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

mpromonet, 2021-01-19:
Hi @ksaye 

This is not linked to arm64 as answered in your question https://github.com/mpromonet/v4l2rtspserver/issues/224#issuecomment-763080302

Best Regards,
Michel.

lagurus, 2021-01-20:
> Hi lagurus,
> 
> It is not really clear what means without success, does is concern v4l2rtspserver ? or v4l2compress ?
> Maybe the logs could help you ?
> 
> Best Regards,
> Michel.

Thanks for answer, I am now happy with mentioned solution using gstreamer. But if you want to look on it, you can check attached log files.

It was started with

`v4l2compress -vv -f H264 /dev/video0 /dev/video4`

and

`./v4l2rtspserver -vv /dev/video4`

I don't see any error in logs, but client (vlc,mplayer) show nothing, but complains:
```
h264 warning: invalid PPS
live555 warning: no data received in 10s. Switching to TCP
h264 warning: invalid PPS
live555 error: no data received in 10s, aborting
```

Many thanks

[v4l2compress.log](https://github.com/mpromonet/v4l2rtspserver/files/5841215/v4l2compress.log)
[v4l2rtspserver.log](https://github.com/mpromonet/v4l2rtspserver/files/5841218/v4l2rtspserver.log)

closed 2022-02-06 by mpromonet