mpromonet/v4l2rtspserver #224: Build failure on Raspberry Pi 3 Model B Plus Rev 1.3: /root/v4l2rtspserver/src/HTTPServer.cpp:78:77: error: no matching function for call to ‘GenericMediaServer::lookupServerMediaSession(const char*&)’

ksaye, 2021-01-15:
I am attempting to build, and it appears that a dependency changed.

```
root@raspberrypi:~# cat /etc/issue
Raspbian GNU/Linux 10 \n \l

root@raspberrypi:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000f4ce13c5
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
```

Step to repro:

1. Install Raspbian
2. clone the repo
3. run cmake . && make

```
[ 95%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 96%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o
/root/v4l2rtspserver/src/HTTPServer.cpp: In member function ‘ServerMediaSubsession* HTTPServer::HTTPClientConnection::getSubsesion(const char*)’:
/root/v4l2rtspserver/src/HTTPServer.cpp:78:77: error: no matching function for call to ‘GenericMediaServer::lookupServerMediaSession(const char*&)’
  ServerMediaSession* session = fOurServer.lookupServerMediaSession(urlSuffix);
                                                                             ^
In file included from /root/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:25,
                 from /root/v4l2rtspserver/src/HTTPServer.cpp:19:
/root/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:48:16: note: candidate: ‘virtual void GenericMediaServer::lookupServerMediaSession(const char*, void (*)(void*, ServerMediaSession*), void*, Boolean)’
   virtual void lookupServerMediaSession(char const* streamName,
                ^~~~~~~~~~~~~~~~~~~~~~~~
/root/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:48:16: note:   candidate expects 4 arguments, 1 provided
/root/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:53:8: note: candidate: ‘void GenericMediaServer::lookupServerMediaSession(const char*, void (GenericMediaServer::*)(ServerMediaSession*))’
   void lookupServerMediaSession(char const* streamName,
        ^~~~~~~~~~~~~~~~~~~~~~~~
/root/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:53:8: note:   candidate expects 2 arguments, 1 provided
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:2663: CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
root@raspberrypi:~/v4l2rtspserver#
```

closed 2021-01-15 by mpromonet

ksaye, 2021-01-16:
Thanks for the update.  It works great and even on a Pi Zero, it consumes very little CPU for 30 FPS at 640 x 480 using the Pi camera.

```
top - 11:42:29 up 5 min,  2 users,  load average: 0.77, 0.55, 0.29
Tasks:  80 total,   2 running,  46 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.0 us, 14.0 sy,  0.0 ni, 78.3 id,  0.0 wa,  0.0 hi,  1.7 si,  0.0 st
KiB Mem :   378120 total,   248068 free,    43304 used,    86748 buff/cache
KiB Swap:   102396 total,   102396 free,        0 used.   283364 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  348 root      20   0   25332  11652  11284 S 13.8  3.1   0:31.39 v4l2rtspserver
  402 root      20   0       0      0      0 I  2.9  0.0   0:06.84 kworker/u2:3-mm
  634 pi        20   0    8244   3280   2708 R  2.2  0.9   0:05.58 top
```

ksaye, 2021-01-19:
@mpromonet , the error seems to have come back:

```
[ 96%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o
In file included from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:24:0:
/home/pi/v4l2rtspserver/inc/HTTPServer.h: In static member function ‘static HTTPServer* HTTPServer::createNew(UsageEnvironment&, Port, UserAuthenticationDatabase*, unsigned int, unsigned int, std::__cxx11::string)’:
/home/pi/v4l2rtspserver/inc/HTTPServer.h:157:48: error: no matching function for call to ‘HTTPServer::setUpOurSocket(UsageEnvironment&, Port&)’
    int ourSocket = setUpOurSocket(env, rtspPort);
                                                ^
In file included from /home/pi/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:25:0,
                 from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:19:
/home/pi/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:91:14: note: candidate: static int GenericMediaServer::setUpOurSocket(UsageEnvironment&, Port&, int)
   static int setUpOurSocket(UsageEnvironment& env, Port& ourPort, int domain);
              ^~~~~~~~~~~~~~
/home/pi/v4l2rtspserver/live/liveMedia/include/GenericMediaServer.hh:91:14: note:   candidate expects 3 arguments, 2 provided
CMakeFiles/libv4l2rtspserver.dir/build.make:4862: recipe for target 'CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o' failed
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/libv4l2rtspserver.dir/all' failed
make[1]: *** [CMakeFiles/libv4l2rtspserver.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
pi@pirtsp:~/v4l2rtspserver $
```

mpromonet, 2021-01-19:
Hi @ksaye 

It is an other breaking change see http://www.live555.com/liveMedia/public/changelog.txt
```
2021.01.18:
- Changed the virtual function "ServerMediaSession::sdpLines()" to take an 'address family'
  parameter.  This lets us ensure (in "OnDemandServerMediaSubsession") that generated SDP
  descriptions for unicast streams have the correct IP address family string ("IP4" or "IP6").
- Changed the "GenericMediaServer::setUpOurSocket()" function to take a 'domain'
  (AF_INET or AF_INET6) parameter, in preparation for eventual support for IPv6.

```

It seems live555 is updating API one by one.... 

Best Regards,
Michel.

ksaye, 2021-01-21:
Yes, it does compile now with just a few warning (on ARM64).  thanks yet again!

commit 339eefdb (2021-01-15): fix #224: build with live 2020.01.14