mpromonet/v4l2rtspserver #225: Starting docker image with audio device

ForgottenTales, 2021-01-16:
I am using the docker image and starting it on a raspberry pi 4:

`docker run --device=/dev/video0 -p 8554:8554 -it mpromonet/v4l2rtspserver`

After that i was able to connect to the video stream from my computer witch 

`vlc rtsp://192.168.178.46:8554/unicast`

However, there is no audio. The help command also does not show the Alsa options, but shouldnt the docker image support it? What am i doing wrong?

ForgottenTales, 2021-01-16:
I realized that when you start the server "normally" you can choose the audio channel via:
`/dev/video0,default `

but i was not able to start the docker image with this option? This seems to be invalid syntax.
`docker run --device=/dev/video0,default -p 8554:8554 -it mpromonet/v4l2rtspserver`

mpromonet, 2021-01-16:
Hi,

There is several problems in what you explain:
- the docker image doesnot have alsa support (if the -h doesnot show audio option it doesnot support it)
- you seems to make confusion between the option of v4l2rtspsever `/dev/video0,default` and the docker option --device, probably what you are looking for is something like `--device=/dev/video0 --device=/dev/snd`

Best Regards,
Michel.

ForgottenTales, 2021-01-16:
Thanks for the quick catch up.

Okay, so it seems I have to compile it by myself when I want audio support? I have a webcam with a built in microphone that I want to stream.

mpromonet, 2021-01-16:
Hi,

There is different possibilities, one could be to use `snap install v4l2-rtspserver`.

Best Regards,
Michel.

ForgottenTales, 2021-01-16:
Thank you so much!

thats Interesting now with the snap installation the audio stream works, but the video disappeared. 

`/snap/v4l2-rtspserver/401/bin/v4l2rtspserver /dev/video0,/dev/video0 -p 8554  -S`

When starting I got this message:

> 021-01-16 21:56:24,206 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/main.cpp:385
        Version: 0.2.1-5-gac37fc3 live555 version:2020.12.13
2021-01-16 21:56:24,208 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/main.cpp:426
        Create V4L2 Source...default
2021-01-16 21:56:24,208 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/v4l2wrapper/src/V4l2MmapDevice.cpp:49
        Device default
VIDIOC_REQBUFS: Inappropriate ioctl for device
2021-01-16 21:56:24,208 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/v4l2wrapper/src/V4l2MmapDevice.cpp:141
        Device default
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
/dev/video0=>default:CARD=U0x46d0x825
2021-01-16 21:56:24,274 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/main.cpp:470
        Create ALSA Source...default:CARD=U0x46d0x825
2021-01-16 21:56:24,275 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/src/ALSACapture.cpp:49
        Open ALSA device: "default:CARD=U0x46d0x825"
2021-01-16 21:56:24,289 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/src/ALSACapture.cpp:117
        set sample format device: default:CARD=U0x46d0x825 to:2 ok
2021-01-16 21:56:24,290 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/src/ALSACapture.cpp:104
        ALSA device: "default:CARD=U0x46d0x825" buffer_size:15052 period_size:940 rate:44100
2021-01-16 21:56:24,290 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/main.cpp:527
        HLS       http://192.168.178.46:8554/ts.m3u8
2021-01-16 21:56:24,290 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/main.cpp:528
        MPEG-DASH http://192.168.178.46:8554/ts.mpd
2021-01-16 21:56:24,290 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/src/DeviceSource.cpp:93
        begin thread
2021-01-16 21:56:24,290 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/inc/V4l2RTSPServer.h:84
        Play this stream using the URL "rtsp://192.168.178.46:8554/unicast"

connecting with vlc to the audio seems. Firefox with the HLS addon is stuck loading.

mpromonet, 2021-01-16:
Hi,

It seems to have a problem with your v4l2 driver. Did you try to reboot or reload the driver.

Best Regards,
Michel.

ForgottenTales, 2021-01-17:
I tried a restart, it does not help. The docker image still works fine (at least the video). So maybe its really a driver issue? What can I try more?

> 2021-01-17 11:00:37,355 [NOTICE] - /build/v4l2-rtspserver/parts/v4l2rtspserver/src/v4l2wrapper/src/V4l2MmapDevice.cpp:141
        Device /dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

ForgottenTales, 2021-01-17:
Okay. I just reinstalled the entire OS and only started a self compiled version of v4l2rtspserver. And now it works fine. Video and Audio! 

So I guess there were some issues with my installation. 

Anyway, thanks a lot for your help! Very usefull tool! I am closing this issue.

closed 2021-01-17 by ForgottenTales