mpromonet/v4l2rtspserver #227: Error During Make

bcwong71, 2021-01-22:
Hello Michael,

I was trying to install the V4l2rstpserver on my Raspberry Pi 4 and found that there was an error during Make. The error message is as follows:
**************************************
_In file included from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:24:
/home/pi/v4l2rtspserver/inc/HTTPServer.h: In constructor ‘HTTPServer::HTTPServer(UsageEnvironment&, int, Port, UserAuthenticationDatabase*, unsigned int, unsigned int, const string&)’:
/home/pi/v4l2rtspserver/inc/HTTPServer.h:170:126: error: no matching function for call to ‘RTSPServer::RTSPServer(UsageEnvironment&, int&, Port&, UserAuthenticationDatabase*&, unsigned int&)’
 Database, reclamationTestSeconds), m_hlsSegment(hlsSegment), m_webroot(webroot)
                                                                               ^

In file included from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:19:
/home/pi/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:101:3: note: candidate: ‘RTSPServer::RTSPServer(UsageEnvironment&, int, int, Port, UserAuthenticationDatabase*, unsigned int)’
   RTSPServer(UsageEnvironment& env,
   ^~~~~~~~~~
/home/pi/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:101:3: note:   candidate expects 6 arguments, 5 provided
In file included from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:19:
/home/pi/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:31:7: note: candidate: ‘constexpr RTSPServer::RTSPServer(const RTSPServer&)’
 class RTSPServer: public GenericMediaServer {
       ^~~~~~~~~~
/home/pi/v4l2rtspserver/live/liveMedia/include/RTSPServer.hh:31:7: note:   candidate expects 1 argument, 5 provided
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:2663: CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2_

******************************************

I remember I was able to install the server successfully on another Raspberry Pi yesterday. However, when I was attempting it again today on a new RPi, the failure message above was prompted and the Make process was unable to continue. Can you help me on this? I made sure that all the dependencies had been installed on both Pis similarly.

My temporary fix was to transfer the V4l2rtspserver folder from the Pi yesterday to the new Pi and the installation took place smoothly without any errors.

Thanks.

Denver

closed 2021-01-22 by mpromonet

bcwong71, 2021-01-22:
Hello, do you know if there is a solution to the problem that I was facing?

mpromonet, 2021-01-22:
Hi,

Maybe `rm -rf live* && cmake . && make` ?

Best Regards,
Michel.

commit ffe99b03 (2021-01-22): fix #227