mpromonet/v4l2rtspserver #232: Raspberry build instructions issue

mickod, 2021-02-06:
Raspberry Pi 4 - version info:
      PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
      NAME="Raspbian GNU/Linux"
      VERSION_ID="10"
      VERSION="10 (buster)"
      VERSION_CODENAME=buster
      Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

Following build instructions here: 

- https://github.com/mpromonet/v4l2rtspserver/wiki/Setup-on-Pi

At the make step the following error is displayed and the make stops:

`[  9%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/live/groupsock/GroupsockHelper.cpp.o
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp: In function ‘void copyAddress(sockaddr_storage&, const sockaddr*)’:
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:859:17: error: ‘const struct sockaddr’ has no member named ‘sa_len’
       if (from->sa_len < sizeof (struct sockaddr_in)) return; // sanity check
                 ^~~~~~
/home/pi/v4l2rtspserver/live/groupsock/GroupsockHelper.cpp:868:17: error: ‘const struct sockaddr’ has no member named ‘sa_len’
       if (from->sa_len < sizeof (struct sockaddr_in6)) return; // sanity check
                 ^~~~~~
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:206: CMakeFiles/libv4l2rtspserver.dir/live/groupsock/GroupsockHelper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
`

Some googling suggest that some linux versions  do not have 'sa_len ' as a member of sockaddr, but as these instructions presumably worked for others n Raspberry releases I am not sure that is to root cause, and if it is what the best way to corret or work around it is.

mpromonet, 2021-02-06:
Hi,

This is because live555 change of interface, and doesnot build on linux with release 2021.02.05. It will probably be fixed soon.
You may build using 
`cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2021.01.29.tar.gz .`

Best Regards,
Michel.

Paul760, 2021-02-06:
Last week there was an error with ‘ourIPAddress’ was not declared and an issue about that was closed without any information. Now 7 days later I tried it for the second time and I get the same error with ‘const struct sockaddr’ has no member named ‘sa_len’...

Using cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2021.01.29.tar.gz . doesnot change anything. On a clean install I try:

sudo apt-get install cmake liblog4cpp5-dev libv4l-dev
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver/
cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2021.01.29.tar.gz .
make

But with make I get the same error with the new cmake -d as with cmake .

mpromonet, 2021-02-06:
Hi @Paul760 

Maybe you did not clean the CMake files, the live folder and the live555.tar.gz ?
I just run your commands : 
```
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver/
cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2021.01.29.tar.gz .
make
```
and it completed.

Best Regards,
Michel.

mickod, 2021-02-06:
Thanks, @mpromonet.

I can confirm the following steps resulted in a a clean build - I first deleted my existing v4l2rtspserver directory:

git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver/
cmake -D LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2021.01.29.tar.gz .
make
sudo make install

If the fix to the regular build is not imminent it might be worth updating the setup instructions with a note about this also?

mpromonet, 2021-02-07:
Hi @mickod 

Usually live555 is fixing quickly this kind of mistake, I guess they are using FreeBSD, and not check if Linux is building, but many users do this and report in the mailing list http://lists.live555.com/pipermail/live-devel/
I will probably add a note in the README, as in the past week, compatibility or build was broken severals times, but this is not common, they seems to work on IPv6 support.... My approach is to follow the latest, I would like to avoid to pin to a version.

Best Regards,
Michel.

mickod, 2021-02-07:
@mpromonet  - makes sense. Thanks.

closed 2021-02-22 by mpromonet