mpromonet/v4l2rtspserver #233: -f format option and codecs

neurogoth, 2021-02-14:
Can you provide here an perhaps update the documentation with clear examples on how the "-f" option can be used ?
Can the v4l2-ctl "fmt" string be passed through it ? I really can't understand what can be passed to that argument.

Also, does this tool support any other codec other than H.264 ?

neurogoth, 2021-02-14:
Ok - I found in other comments that "-f" keeps the existing format set on the device. I think this could be stated more clearly - and also "-f<format>" ... can we get an example of input for <format> in that use of the arg.

mpromonet, 2021-02-21:
Hi @neurogoth 

This is describe in the README and in the -h : 
```
	 -f       : V4L2 capture using current capture format (-W,-H are ignored)
	 -fformat : V4L2 capture using format (-W,-H are used)
```

Feel free to submit a clearer description using a Pull Request.

About the argument, there is no format limitation, the argument is the fourcc, then it could be H264, VP90, JPEG,.... 

Best Regards,
Michel.

closed 2021-02-28 by mpromonet