mpromonet/v4l2rtspserver #234: feature request: create a rtspserver class to use it in other customized projects

jgomezlopez, 2021-02-18:
It is handy to have a general class to use in other projects use rtsp server functionality

mpromonet, 2021-02-21:
Dear @jgomezlopez 

This is the aim of [V4l2RTSPServer](https://github.com/mpromonet/v4l2rtspserver/blob/master/inc/V4l2RTSPServer.h), one example that use it is https://github.com/mpromonet/v4l2onvif/blob/master/src/onvif-server.cpp 

Best Regards,
Michel.

jgomezlopez, 2021-02-22:
I was thinking in converting the main functionality of the main file in a  class.

mpromonet, 2021-02-22:
Hi @jgomezlopez 

This is what does the class V4l2RTSPServer, and the library libv4l2rtspserver, that I am using in others projects like https://github.com/mpromonet/v4l2onvif and https://github.com/mpromonet/v4l2web.
Feel free to submit a Pull Request to precise what you have in mind.

Best Regards,
Michel.

closed 2021-02-22 by mpromonet