mpromonet/v4l2rtspserver #236: fatal error: liveMedia.hh: No such file or directory

flatsiedatsie, 2021-02-24:
I was following the proces in the readme to create the project. I was hoping to run it on a Pi Zero, and use it as a smart doorbell. However, during the `make` process on a Raspberry Pi Zero I ran into this issue:

```
Scanning dependencies of target v4l2wrapper
[  4%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Access.cpp.o
[  9%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Capture.cpp.o
[ 14%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Device.cpp.o
[ 19%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2MmapDevice.cpp.o
[ 23%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Output.cpp.o
[ 28%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2ReadWriteDevice.cpp.o
[ 33%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/logger.cpp.o
[ 38%] Linking CXX static library libv4l2wrapper.a
[ 38%] Built target v4l2wrapper
Scanning dependencies of target libv4l2rtspserver
[ 42%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 47%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/DeviceSource.cpp.o
In file included from /home/pi/v4l2rtspserver/src/DeviceSource.cpp:18:
/home/pi/v4l2rtspserver/inc/DeviceSource.h:24:10: fatal error: liveMedia.hh: No such file or directory
 #include <liveMedia.hh>
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:76: CMakeFiles/libv4l2rtspserver.dir/src/DeviceSource.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

flatsiedatsie, 2021-02-24:
I downloaded the latest version of live5555 and manually copied files into the `inc` folder. After a few tries it completed, although with an error:

```
MJPEGVideoSource.cpp:(.text+0xc8c): undefined reference to `FramedSource::afterGetting(FramedSource*)'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x130): undefined reference to `MediaSource::isSource() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x134): undefined reference to `Medium::isSink() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x138): undefined reference to `Medium::isRTCPInstance() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x13c): undefined reference to `Medium::isRTSPClient() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x140): undefined reference to `Medium::isRTSPServer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x144): undefined reference to `Medium::isMediaSession() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x148): undefined reference to `Medium::isServerMediaSession() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x154): undefined reference to `MediaSource::getAttributes() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x158): undefined reference to `MediaSource::MIMEtype() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x15c): undefined reference to `FramedSource::isFramedSource() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x160): undefined reference to `MediaSource::isRTPSource() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x164): undefined reference to `MediaSource::isMPEG1or2VideoStreamFramer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x168): undefined reference to `MediaSource::isMPEG4VideoStreamFramer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x16c): undefined reference to `MediaSource::isH264VideoStreamFramer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x170): undefined reference to `MediaSource::isH265VideoStreamFramer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x174): undefined reference to `MediaSource::isDVVideoStreamFramer() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x178): undefined reference to `JPEGVideoSource::isJPEGVideoSource() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x17c): undefined reference to `MediaSource::isAMRAudioSource() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x180): undefined reference to `MediaSource::isMPEG2TransportStreamMultiplexor() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x184): undefined reference to `FramedSource::maxFrameSize() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x1b0): undefined reference to `typeinfo for JPEGVideoSource'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o): in function `MJPEGVideoSource::doGetNextFrame()':
MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSource14doGetNextFrameEv[_ZN16MJPEGVideoSource14doGetNextFrameEv]+0x54): undefined reference to `FramedSource::getNextFrame(unsigned char*, unsigned int, void (*)(void*, unsigned int, unsigned int, timeval, unsigned int), void*, void (*)(void*), void*)'
/usr/bin/ld: MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSource14doGetNextFrameEv[_ZN16MJPEGVideoSource14doGetNextFrameEv]+0x64): undefined reference to `FramedSource::handleClosure(void*)'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o): in function `MJPEGVideoSource::doStopGettingFrames()':
MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSource19doStopGettingFramesEv[_ZN16MJPEGVideoSource19doStopGettingFramesEv]+0x18): undefined reference to `FramedSource::doStopGettingFrames()'
/usr/bin/ld: MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSource19doStopGettingFramesEv[_ZN16MJPEGVideoSource19doStopGettingFramesEv]+0x38): undefined reference to `FramedSource::stopGettingFrames()'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o): in function `MJPEGVideoSource::~MJPEGVideoSource()':
MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSourceD2Ev[_ZN16MJPEGVideoSourceD5Ev]+0x28): undefined reference to `Medium::close(Medium*)'
/usr/bin/ld: MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSourceD2Ev[_ZN16MJPEGVideoSourceD5Ev]+0x34): undefined reference to `JPEGVideoSource::~JPEGVideoSource()'
/usr/bin/ld: liblibv4l2rtspserver.a(MemoryBufferSink.cpp.o): in function `MemoryBufferSink::MemoryBufferSink(UsageEnvironment&, unsigned int, unsigned int, unsigned int)':
MemoryBufferSink.cpp:(.text+0x28): undefined reference to `MediaSink::MediaSink(UsageEnvironment&)'
/usr/bin/ld: MemoryBufferSink.cpp:(.text+0xb8): undefined reference to `MediaSink::~MediaSink()'
/usr/bin/ld: liblibv4l2rtspserver.a(MemoryBufferSink.cpp.o): in function `MemoryBufferSink::~MemoryBufferSink()':
MemoryBufferSink.cpp:(.text+0x124): undefined reference to `MediaSink::~MediaSink()'
/usr/bin/ld: liblibv4l2rtspserver.a(MemoryBufferSink.cpp.o): in function `MemoryBufferSink::continuePlaying()':
[... cut ...]
/usr/bin/ld: liblibv4l2rtspserver.a(MemoryBufferSink.cpp.o):(.rodata+0xa0): undefined reference to `MediaSink::sourceIsCompatibleWithUs(MediaSource&)'
/usr/bin/ld: liblibv4l2rtspserver.a(MemoryBufferSink.cpp.o):(.rodata+0xb0): undefined reference to `typeinfo for MediaSink'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:87: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

mpromonet, 2021-02-24:
Hi,

You probably did some mix, try something like : 
```
git clean -x -f -d
cmake . 
make 
```

Best Regards,
Michel.

flatsiedatsie, 2021-02-25:
That removes all the files I added, and brings me back the start.

The only files that were missing were from the `live5555` project. It's the only project I added files from.

Should I use an older version of live555 perhaps?

mpromonet, 2021-02-25:
Hi,

Normally cmake command download live555.
You should have something like : 
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1")
-- CMAKE_TOOLCHAIN_FILE=
-- CMAKE_C_COMPILER=/usr/bin/cc
-- CMAKE_CXX_COMPILER=/usr/bin/c++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
running git submodule update --init 2>&1
x live/
x live/liveMedia/
x live/groupsock/
x live/UsageEnvironment/
x live/BasicUsageEnvironment/
x live/testProgs/
x live/mediaServer/
x live/proxyServer/
x live/hlsProxy/
x live/WindowsAudioInputDevice/
x live/configure
x live/config.armeb-uclibc
x live/config.armlinux
x live/config.avr32-linux
....
```

I just made a try on a raspberry zero. It build slowly, but it builds....

Best Regards,
Michel.

closed 2021-02-28 by mpromonet