mpromonet/v4l2rtspserver #237: Image garbled at high resolution

camaro4life18, 2021-02-25:
When I set the resolution for the stream at the camera maximum, it comes out all garbled.
Here is the supported formats for my camera:
`v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 3264x2448
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 2592x1944
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 2048x1536
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1600x1200
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.050s (20.000 fps)
        [1]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 3264x2448
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 2592x1944
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 2048x1536
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1600x1200
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
`

Then when I run the command...
v4l2rtspserver -W 3264 -H 2448 -F 15 -P 8554 -f MJPEG /dev/video0

This image looks distorted and the resolution on the client comes back as 1216x400

[image]

camaro4life18, 2021-02-26:
Digging into my own issue, I ran with -vv and got the follow logs...

> 2021-02-26 12:18:02,014 [DEBUG ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:26
>         SOF length:17
> 2021-02-26 12:18:02,014 [INFO  ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:42
>         width:1216 height:400 type:0 precision:8
> 2021-02-26 12:18:02,014 [DEBUG ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:49
>         DQT length:132
> 2021-02-26 12:18:02,014 [DEBUG ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
>         Quantization table idx:0 precision:0 size:64 total size:64
> 2021-02-26 12:18:02,014 [DEBUG ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:77
>         SOS length:12
> 2021-02-26 12:18:02,014 [DEBUG ] - /home/pi/v4l2rtspserver/src/MJPEGVideoSource.cpp:87
>         headerSize:589
> 2021-02-26 12:18:02,018 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:105
>         waitingFrame    delay:66ms
> 2021-02-26 12:18:02,019 [INFO  ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:29
>         intv_sec:1614359882 fps:15 bandwidth:45002kbps
> 2021-02-26 12:18:02,019 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:213
>         getNextFrame    timestamp:1614359882.18177      size:384016     diff:1ms
> 2021-02-26 12:18:02,019 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:245
>         queueFrame      timestamp:1614359882.18177      size:384016     diff:1ms
> 2021-02-26 12:18:02,084 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:105
>         waitingFrame    delay:66ms
> 2021-02-26 12:18:02,085 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:213
>         getNextFrame    timestamp:1614359882.84836      size:383872     diff:1ms
> 2021-02-26 12:18:02,085 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:245
>         queueFrame      timestamp:1614359882.84836      size:383872     diff:1ms
> 2021-02-26 12:18:02,139 [DEBUG ] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:162
>         deliverFrame    timestamp:1614359882.139887     size:384016     diff:121ms      queue:1

In those logs you can see the resolution is getting shrunk. I dug in the code to see where that output is coming from and traced some of the code back to DeviceSource.cpp line 150 that is reducing the size to fit in fMaxSize. So it looks like live555 is reducing the image size to fit its buffer.

I'm not familiar with this code and how live555 and ffmpeg work, but this definitely seems like the problem for such a high resolution. It also looks like i'm not the only one to ever run into this issue....
https://stackoverflow.com/questions/20510484/live555-fmaxsize-and-ffmpeg

mpromonet, 2021-02-26:
Hi camaro4life18,

This project doesnot use ffmpeg, it doesnot compress images. The MJEG is not implemented in live555 and this project brings a naive implementation of it. Depending on the JPEG image, it may work or not.
It seems size if not correctly decode, you should look to your JPEG frame regarding https://github.com/mpromonet/v4l2rtspserver/blob/master/src/MJPEGVideoSource.cpp

Best Regards,
Michel.

Syntaxrabbit, 2025-02-09:
I just stumpled upon this issue as that's exactly my problem at the moment.
I need to stream 4K from a raspberry pi to a PC, and so far, this project is the only solution that runs stable and near "real time" without killing cpu and ram for unnecessary re-encoding.
Any chance to get something like a workaround for this? At the moment 2048x1536 is all I can get.
Or are there any alternative projects at all?

mpromonet, 2025-02-09:
Hi @Syntaxrabbit 

The limit of 2048 is for JPEG format, with others format it should works 

Best regards 
Michel

Syntaxrabbit, 2025-02-09:
Unfortunately the camera only delivers MJPEG and transcoding on a raspberry pi is really painful.

mpromonet, 2025-02-11:
Hi @Syntaxrabbit 
Raspberry have an h264 encoder quite easy to use. 
Maybe the rtp for JPEG has evolved...
Best regards 
Michel