mpromonet/v4l2rtspserver #240: Stream freeze often

azsde, 2021-03-09:
Hello,

I have an issue with the current master build, the stream freeze often for 10-15 seconds, and will then resume, skipping over the 15 seconds time frame.

This is not ideal because everything that happens during these 10-15 seconds are lost.

Where can I find logs that might help me out ?

mpromonet, 2021-03-09:
Hi @azsde 

It may be network that drop frames.
One approach is to use wireshark (or similar tools), another is to make connection through RTP over RTSP (to see if loosing frame becomes retransmission delay)

Best Regards,
Michel.

mpromonet, 2021-03-09:
Hi Azsde,

In order to ask for TCP stream, you should send an RTSP SETUP, this could be done using `--rtsp-tcp` of vlc or `-rtsp_transport tcp` of ffmpeg 

Best Regards,
Michel.

azsde, 2021-03-09:
I'll try this out, the thing is that the client I use is Synology's Surveillance Station, I do not have control over how it gets it's content from the raspberry :(