mpromonet/v4l2rtspserver #241: Is there a way to setup the server to use multistreams?

D0T-C0M, 2021-03-09:
I'm looking for a way to have 2 streams streaming simultaneously everything identical except one would be an HD resolution stream and one at a lower resolution, everything else framerate, I-frame etc would stay the same.   Is this possible?

D0T-C0M, 2021-03-12:
How would you specify a different resolution for each /dev/video0 and /dev/device1 ?  

Also the how do you change the url like rtsp://xxx.xxx.xxx.xxx:8554/unicast for device0 and rtsp://xxx.xxx.xxx.xxx:8854/unicast2 for device1? 

sorry I'm a dummy.  Thank you for your time.

D0T-C0M, 2021-03-13:
I tried to figure this out but how do I specify different resolutions for each device0 and device1?

mpromonet, 2021-04-04:
Hi @D0T-C0M ,

This depends on the capture layer, this program consume V4L2 frames and provide RTSP streams. In order to have 2 V4L2 device with different resolution, this depend on it. You may look to v4l2loopback project that could be entry of v4l2rtspserver.

Best Regards,
Michel.

CyanAutomation, 2021-11-09:
Yes, a lower resolution sub stream would be a great addition. 

Frigate (which is rapidly gaining popularity), works best with a low resolution + FPS stream for detection, and a HQ stream for recording.

closed 2022-02-06 by mpromonet