mpromonet/v4l2rtspserver #244: Cannot compile on Raspi 4B

jwpleow, 2021-04-05:
Hello,

I'm on a fresh install of Raspberry Pi 4B   5.4.51-v7l+

I first installed live555 like this
```
wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz
tar xvzf live555-latest.tar.gz
cd live
./genMakefiles linux
make
sudo make install
```

Installed dependencies
```
sudo apt install -y liblivemedia-dev liblog4cpp5-dev libasound2-dev
```
and then proceeded to follow your installation steps.

While running `make` I'm getting the linking error
```
[ 95%] Building CXX object CMakeFiles/v4l2rtspserver.dir/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o: in function `HTTPServer::HTTPClientConnection::HTTPClientConnection(RTSPServer&, int, sockaddr_storage const&)':
main.cpp:(.text._ZN10HTTPServer20HTTPClientConnectionC2ER10RTSPServeriRK16sockaddr_storage[_ZN10HTTPServer20HTTPClientConnectionC5ER10RTSPServeriRK16sockaddr_storage]+0x2c): undefined reference to `RTSPServer::RTSPClientConnection::RTSPClientConnection(RTSPServer&, int, sockaddr_storage const&)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o: in function `HTTPServer::createNew(UsageEnvironment&, Port, UserAuthenticationDatabase*, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
main.cpp:(.text._ZN10HTTPServer9createNewER16UsageEnvironment4PortP26UserAuthenticationDatabasejjNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10HTTPServer9createNewER16UsageEnvironment4PortP26UserAuthenticationDatabasejjNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x34): undefined reference to `GenericMediaServer::setUpOurSocket(UsageEnvironment&, Port&, int)'
/usr/bin/ld: main.cpp:(.text._ZN10HTTPServer9createNewER16UsageEnvironment4PortP26UserAuthenticationDatabasejjNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10HTTPServer9createNewER16UsageEnvironment4PortP26UserAuthenticationDatabasejjNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x50): undefined reference to `GenericMediaServer::setUpOurSocket(UsageEnvironment&, Port&, int)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o: in function `HTTPServer::HTTPServer(UsageEnvironment&, int, int, Port, UserAuthenticationDatabase*, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
main.cpp:(.text._ZN10HTTPServerC2ER16UsageEnvironmentii4PortP26UserAuthenticationDatabasejjRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN10HTTPServerC5ER16UsageEnvironmentii4PortP26UserAuthenticationDatabasejjRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x44): undefined reference to `RTSPServer::RTSPServer(UsageEnvironment&, int, int, Port, UserAuthenticationDatabase*, unsigned int)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o: in function `V4l2RTSPServer::addSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::list<ServerMediaSubsession*, std::allocator<ServerMediaSubsession*> > const&)':
main.cpp:(.text._ZN14V4l2RTSPServer10addSessionERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS0_4listIP21ServerMediaSubsessionSaISA_EEE[_ZN14V4l2RTSPServer10addSessionERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS0_4listIP21ServerMediaSubsessionSaISA_EEE]+0x134): undefined reference to `RTSPServer::rtspURL(ServerMediaSession const*, int, unsigned char) const'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o: in function `V4l2RTSPServer::AddHlsSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, StreamReplicator*, StreamReplicator*)':
main.cpp:(.text._ZN14V4l2RTSPServer13AddHlsSessionERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiP16StreamReplicatorS9_[_ZN14V4l2RTSPServer13AddHlsSessionERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiP16StreamReplicatorS9_]+0x94): undefined reference to `ourIPv4Address(UsageEnvironment&)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x2c): undefined reference to `GenericMediaServer::lookupServerMediaSession(char const*, void (*)(void*, ServerMediaSession*), void*, unsigned char)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x54): undefined reference to `RTSPServer::specialClientAccessCheck(int, sockaddr_storage const&, char const*)'
/usr/bin/ld: CMakeFiles/v4l2rtspserver.dir/main.cpp.o:(.rodata._ZTV10HTTPServer[_ZTV10HTTPServer]+0x58): undefined reference to `RTSPServer::specialClientUserAccessCheck(int, sockaddr_storage const&, char const*, char const*)'
/usr/bin/ld: liblibv4l2rtspserver.a(HTTPServer.cpp.o):(.rodata+0x550): undefined reference to `RTSPServer::RTSPClientConnection::handleCmd_DESCRIBE_afterLookup(ServerMediaSession*)'
/usr/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createRtpSink(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0x138): undefined reference to `Groupsock::Groupsock(UsageEnvironment&, sockaddr_storage const&, Port, unsigned char)'
/usr/bin/ld: MulticastServerMediaSubsession.cpp:(.text+0x1ec): undefined reference to `Groupsock::Groupsock(UsageEnvironment&, sockaddr_storage const&, Port, unsigned char)'
/usr/bin/ld: MulticastServerMediaSubsession.cpp:(.text+0x22c): undefined reference to `RTCPInstance::createNew(UsageEnvironment&, Groupsock*, unsigned int, unsigned char const*, RTPSink*, RTPSource*, unsigned char, SRTPCryptographicContext*)'
/usr/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::sdpLines()':
MulticastServerMediaSubsession.cpp:(.text+0x2d0): undefined reference to `PassiveServerMediaSubsession::sdpLines(int)'
/usr/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.rodata+0x120): undefined reference to `PassiveServerMediaSubsession::sdpLines(int)'
/usr/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.rodata+0x124): undefined reference to `PassiveServerMediaSubsession::getStreamParameters(unsigned int, sockaddr_storage const&, Port const&, Port const&, int, unsigned char, unsigned char, sockaddr_storage&, unsigned char&, unsigned char&, Port&, Port&, void*&)'
/usr/bin/ld: liblibv4l2rtspserver.a(ServerMediaSubsession.cpp.o): in function `BaseServerMediaSubsession::createSource(UsageEnvironment&, FramedSource*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
ServerMediaSubsession.cpp:(.text+0x78): undefined reference to `H264VideoStreamDiscreteFramer::createNew(UsageEnvironment&, FramedSource*, unsigned char, unsigned char)'
/usr/bin/ld: ServerMediaSubsession.cpp:(.text+0xb0): undefined reference to `H265VideoStreamDiscreteFramer::createNew(UsageEnvironment&, FramedSource*, unsigned char, unsigned char)'
/usr/bin/ld: liblibv4l2rtspserver.a(TSServerMediaSubsession.cpp.o):(.rodata+0x224): undefined reference to `OnDemandServerMediaSubsession::sdpLines(int)'
/usr/bin/ld: liblibv4l2rtspserver.a(TSServerMediaSubsession.cpp.o):(.rodata+0x228): undefined reference to `OnDemandServerMediaSubsession::getStreamParameters(unsigned int, sockaddr_storage const&, Port const&, Port const&, int, unsigned char, unsigned char, sockaddr_storage&, unsigned char&, unsigned char&, Port&, Port&, void*&)'
/usr/bin/ld: liblibv4l2rtspserver.a(TSServerMediaSubsession.cpp.o):(.rodata+0x280): undefined reference to `OnDemandServerMediaSubsession::createGroupsock(sockaddr_storage const&, Port)'
/usr/bin/ld: liblibv4l2rtspserver.a(TSServerMediaSubsession.cpp.o):(.rodata._ZTV19AddH26xMarkerFilter[_ZTV19AddH26xMarkerFilter]+0x58): undefined reference to `MediaSource::isMPEG2TransportStreamMultiplexor() const'
/usr/bin/ld: liblibv4l2rtspserver.a(UnicastServerMediaSubsession.cpp.o):(.rodata+0x120): undefined reference to `OnDemandServerMediaSubsession::sdpLines(int)'
/usr/bin/ld: liblibv4l2rtspserver.a(UnicastServerMediaSubsession.cpp.o):(.rodata+0x124): undefined reference to `OnDemandServerMediaSubsession::getStreamParameters(unsigned int, sockaddr_storage const&, Port const&, Port const&, int, unsigned char, unsigned char, sockaddr_storage&, unsigned char&, unsigned char&, Port&, Port&, void*&)'
[... cut ...]
/usr/bin/ld: liblibv4l2rtspserver.a(H264_V4l2DeviceSource.cpp.o):(.rodata+0x1e0): undefined reference to `MediaSource::isMPEG2TransportStreamMultiplexor() const'
/usr/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.rodata+0x188): undefined reference to `MediaSource::isMPEG2TransportStreamMultiplexor() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:88: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
```

I notice live555 release a new version recently (22nd March) - could this be related?

mpromonet, 2021-04-05:
Hi,

It seems you mixed live555 include & library.
If you don't do any live555 install, running cmake will download & built it.
If live555 package is detected by cmake, it will use it.
I guess in your way to go, the problem is that you install the latest release that was used for builing and the liblivemedia-dev is used for linking.
Without installing your own live555 have better change to work. Otherwise you should look to the CMakefile why this conflict occurs. Clearly this case is not forseen.

Best Regards,
Michel.

BlueBayPhil, 2021-04-11:
Have you managed to resolve this? Currently running into the exact same issue =/
Tried looking for a way to uninstall the version of live555 I installed but there doesn't seem to be an easy way to do it

BlueBayPhil, 2021-04-11:
I managed to get this to compile in the end by editing the CMakeLists.txt file. Under the section `#live555` remove the if statement that checks for the live555 stuff and force it to always download live555.

```
# live555
find_package(LiveMedia)
find_package(UsageEnvironment)
find_package(BasicUsageEnvironment)
find_package(GroupSock)
set(LIVE ${CMAKE_BINARY_DIR}/live)
set(LIVEINCLUDE ${LIVE}/groupsock/include ${LIVE}/liveMedia/include ${LIVE}/UsageEnvironment/include ${LIVE}/BasicUsageEnvironment/include)
if (NOT EXISTS ${LIVE})
    file (DOWNLOAD ${LIVE555URL} ${CMAKE_BINARY_DIR}/live555-latest.tar.gz )
    EXECUTE_PROCESS(COMMAND ${CMAKE_COMMAND} -E tar xvf ${CMAKE_BINARY_DIR}/live555-latest.tar.gz  RESULT_VARIABLE unpack_result)
    if(NOT unpack_result STREQUAL "0")
        message(FATAL_ERROR "Fetching and compiling live555 failed!")
    endif()
endif()
```

This is probably not the most elegant or best way to resolve this in the long run but it gets it to compile and install.

mpromonet, 2021-04-11:
Hi @BlueBayPhil 

Did you try to uninstall the live555 version you installed ?

Best Regards,
Michel.

BlueBayPhil, 2021-04-12:
Hi @mpromonet I uninstalled the version of live555 before trying my "solution" and it diddn't seem to resolve the problem.

As I am running on a debian distro (specifially raspbian buster) I created a .deb package from the installer  to aide in the uninstallation so it may have left some gunk behind.

mpromonet, 2021-04-12:
Hi @BlueBayPhil 

You are right, the live555 source download by the CMakefile.txt are uncompressed inside the source, usually in `CMakeFiles/libv4l2rtspserver.dir/live`. The PR https://github.com/mpromonet/v4l2rtspserver/pull/238 has changed a little the way to look for live555 using the system package if present.

As I understand, it may needs some improvement to support a manual make install of live555.

Feel free to submit a PR to improve this.

Best Regards,
Michel.

closed 2021-07-04 by mpromonet