mpromonet/v4l2rtspserver #245: USB webcam glitches , Pi camera does not

sithdaddy, 2021-04-05:
Hello.
I am experimenting with v4l2rtspserver and a logitech c920 webcam and a Pi camera. When i stream with the PI camera module the stream is fine over vlc , no issues at all.
When i use the Logitech webcam and stream , and watch the stream on vlc the image smudges and turns grey every few seconds.

This issue is the same over wifi and ethernet.

Any idea what might be the issue ?

my setup:

Pi zero w
Pi camera
logitech c920 webcam

Taudris, 2021-08-31:
I'm having the same issue on a 3B+ and C920. I don't have a Pi/CSI camera to try.

mpromonet, 2021-08-31:
Hi,

You may try to reduce de resolution or to use RTP over TCP.

Best Regards,
Michel.

fplanque, 2024-03-23:
I have the same issue and I can provide some additional details:

- Raspi camera module works flawlessly (H264)
- Some cheap USB camera module from AliExpress works flawlessly (MJPEG)
- Logitech C270 (cannot do H264, so MJPEG only): 
  - Works in VLC in MJPG
  - Green image with some movement in BlueIris (MJPG)
- Logitech C920 (can do MJPG and H264):
  - MJPG:
    - VLC displays image but colors are way off (like vivid solarized effect) 
    - BlueIris displays Green image with some movement 
  - H264:
    - VLC displays the bottom 25% of the image correctly (top is garbled)
    - BlueIris displays the bottom 50% correctly (top is green)

I wonder if the issue is specific to Logitech... :?

I'd like to understand what is going on. I am particularly interested in BlueIris in order to record the streams.

Michel:
> You may try to reduce de resolution or to use RTP over TCP.

I did try to set the resolution of the C920 all the way down to 160x120. It did not many any difference (except the bottom 25% of the picture that displays is now very pixellated ;)

I also tried to reduce the framerate to 10fps. No change.

Can you please elaborate on what you mean with "use RTP over TCP" ?  (I used "rtsp://" and I don't know what RTP over TCP would look like)

Thank you for this project and all the time you put into it! :)

mpromonet, 2024-03-24:
Hi @fplanque 

RTP over TCP negociation can be set using `--rtsp-tcp`  argument of VLC.

Best Regards,
Michel.

sithdaddy, 2024-03-24:
Tried to use v4l2-ctl and change the output format of the cam?
I think those logitech cams can put out a h264 stream which the rtsp server does not like.

this is what i had to do to make it work...

```
sleep 10

sudo rmmod v4l2loopback &

sleep 1

sudo modprobe v4l2loopback video_nr=20 &

sleep 2

sudo v4l2-ctl -d /dev/video0 --set-fmt-video=width=800,height=448,pixelformat=YUYV &

sleep 1

sudo v4l2-ctl --set-ctrl focus_auto=0 -d /dev/video0 &

sleep 1

sudo v4l2-ctl --set-ctrl focus_absolute=44 -d /dev/video0 &

sleep 1

sudo v4l2-ctl --set-ctrl sharpness=255 -d /dev/video0 &

sleep 1

sudo v4l2compress_omx /dev/video0 /dev/video20 &

```