mpromonet/v4l2rtspserver #246: Change IP adress and latency question

TheKerry, 2021-04-14:
My Raspberry pi is in a VPN network with Zerotier. I would like to make the stream available under a different IP address, namely the one in the Zerotier VPN. Is there a way to do this?
Second question: Which settings result in the lowest latency?

mpromonet, 2021-04-14:
Hi @TheKerry 

The latency depends on capture, compressing, network transfert and bufferization, uncompressing, display.
You probably have lowest latency using small resolution and powerful network. 
If you are using VLC, you may try to reduce the `--network-caching` value

Best Regards,
Michel.

mpromonet, 2022-02-06:
Hi,

Using
```
 vlc --network-caching=100 rtsp://127.0.0.1:8554/unicast
 ffplay -probesize 32 rtsp://127.0.0.1:8554/unicast

```
reduce latency to about 500ms (capturing a counter on the screen)
With default parameters, it is about 1~2s.

Best Regards,
Michel.

closed 2022-02-06 by mpromonet