mpromonet/v4l2rtspserver #249: Unable to change port to 80

sylver9, 2021-04-22:
I would like to launch the stream on port 80 but it doesn't allow me. Any ideas why and how to do it on port 80.
`v4l2rtspserver -W 640 -H 480 -F 15 -P 8554 /dev/video0`

Thanks.

mpromonet, 2021-04-22:
Hi,

Usually a normal user cannot open a port below 1024.
You may look at https://serverfault.com/questions/112795/how-to-run-a-server-on-port-80-as-a-normal-user-on-linux 

Best Regards,
Michel.

sylver9, 2021-04-23:
Hi Michel,
Thanks for sharing the link. 
The reason I want to launch it on port 80 is that the default port 8554 or other ports like 8181 seem to be working but I'm unable to stream RTSP to internet even though it works in my LAN. I have done port forwarding did all basic troubleshooting but cannot understand what's wrong. Even tried it in a different networking using a higher end router but the problem remains -rtsp works in LAN but not on WAN even with router's port forwarding configured. Is this a problem with the RTSP stream on the raspberry? Any advise would be nice. 

Thanks.

mpromonet, 2021-04-23:
Hi sylver9,

By default the RTSP server use RTP/UDP, this probably not corss your router.
A possible approach is to use RTP over TCP (sending RTSP DESCRIBE with TCP), or using RTSP over HTTP (you can enable this with the -p argument)

Best Regards,
Michel.

sylver9, 2021-04-24:
Thanks Michel.
I tried using -p option but I think I don't understand it correctly. Even when I used the command with -p 8181 as below it opens the stream on 8554. Can you please clarify "-p". Is it to be used in same way as "-P"?
`v4l2rtspserver -W 640 -H 480 -F 15 -p 8181 /dev/video0`
the message at the bottom says use the link rtsp://19.....134:8554/unicast

I'm able to launch the stream on port 80 using sudo but it didn't help. The stream still cannot be reached from internet (WAN) although it works in LAN. I configured my router to forward TCP & UDP ports to the ip address of the raspberry. See screenshot below. Any insights would be helpful. Thanks.

[image]

mpromonet, 2021-04-25:
Hi sylver9,

Using `-p 8181` enable a new port for RTSP over HTTP tunneling, RTSP & RTP will go to this unique port.
For instance this article explain how to set option in VLC to configure the live555 demuxer https://mynuuo.zendesk.com/hc/en-us/articles/360004120753-How-to-setup-and-configure-VLC-player-to-retrieve-camera-RTSP-streaming-
From my memory adding a port enable the feature RTSP over HTTP, then you can use both port without differences 8181 or 8554.
Also setting RTP over TCP cross the router, the routing problem with UDP, is the SDP publish dynamic port (UDP port is not 8554 when RTSP use 8554 port) that are not reachable from the WAN (or you need to define a range of port).

Best Regards,
Michel.