mpromonet/v4l2rtspserver #250: Has anyone managed to get this to run via docker-compose?

kllngtme, 2021-05-04:
@sndnvaps Did you do something else to get it working with docker-compose?
When using the command like you it doesn't work at all. I need to write it in the `["", ""]` form. That way the container starts successfully but I still can't access the stream. When doing the same settings via normal docker it works. I've also tried stdin_open and tty true. No success either.

_Originally posted by @konstantinj in https://github.com/mpromonet/v4l2rtspserver/issues/157#issuecomment-583736161_

kllngtme, 2021-05-04:
I'm having the same issue essentially. Running this simply works but if I try to put it in a docker compose I get issues with it starting.. It comes up with an error: `v4l2rtspserver    | log level:500`

Running with this command works no problems:
`docker run --device=/dev/video0 -p 8554:8554 -it mpromonet/v4l2rtspserver`

If using a simple docker-compose of the following I get the error:
```
  v4l2rtspserver:
    restart: unless-stopped
    image: mpromonet/v4l2rtspserver
    container_name: v4l2rtspserver
    ports:
      - "8554:8554"
    devices:
      - "/dev/video0:/dev/video0"
```

mpromonet, 2021-05-04:
Hi kllngtme,

Using docker-compose should work the same as docker, may be the error will help you to understand the problem.
By the way ```v4l2rtspserver | log level:500``` is not an error it is the output log.
I made a quick test with 
```
version: "3.5"
services:
  v4l2rtspserver:
    image: mpromonet/v4l2rtspserver
    ports:
      - "8554:8554"
    devices:
      - "/dev/video0:/dev/video0"
    command: ["-G","640x480x25", "-S"]
```
It seems to run normally, I am able to reach the HLS stream using http://pi2.local:8554/

Best Regards,
Michel.

sndnvaps, 2021-05-06:
@kllngtme i had drop it .it not work for my usb webcam in rpi

mpromonet, 2021-05-06:
Hi @sndnvaps,

It needs a V4L2 device that support a compressed format (for instance H264, JPEG), like the raspicam.
With a webcam that supports YUV/RGB format, you may involve https://github.com/mpromonet/v4l2tools that can compress to H264 format using OMX.

Best Regards,
Michel.

kllngtme, 2021-05-07:
Ah yes, I see now. It does start, I guess I should've tested it first. I didn't see any notice of anything starting up other than that log level notice. When starting it with docker alone it spits out all this:

```
[NOTICE] /v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] /v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] /v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
        /dev/video0 support streaming
[NOTICE] /v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
        /dev/video0:H264 size:1024x768 bufferSize:786432
```
 Either way, good to go. Thank you

closed 2021-07-04 by mpromonet