mpromonet/v4l2rtspserver #251: Connection Error

hedjour, 2021-06-11:
Sorry if it's no a v4l2rtspserver error
I have installed your software on a rpi whithout internet
```bash
systemctl status v4l2rtspserver.service 
● v4l2rtspserver.service - V4L2 RTSP server
   Loaded: loaded (/lib/systemd/system/v4l2rtspserver.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-06-10 16:48:07 CEST; 20h ago
 Main PID: 338 (v4l2rtspserver)
    Tasks: 2 (limit: 725)
   CGroup: /system.slice/v4l2rtspserver.service
           └─338 /usr/local/bin/v4l2rtspserver -H 1920 -W 1080 -F 30 -P 8888 /dev/video0

juin 10 16:48:08 raspberrypi v4l2rtspserver[338]: 2021-06-10 16:48:08,513 [NOTICE] - /home/pi/v4l2rtspse
juin 10 16:48:08 raspberrypi v4l2rtspserver[338]:         Device /dev/video0 nb buffer:10
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]: 2021-06-10 16:48:09,115 [NOTICE] - /home/pi/v4l2rtspse
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]:         Create Source .../dev/video0
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]: 2021-06-10 16:48:09,123 [NOTICE] - /home/pi/v4l2rtspse
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]:         begin thread
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]: 2021-06-10 16:48:09,130 [NOTICE] - /home/pi/v4l2rtspse
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]:         format:video/H264
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]: 2021-06-10 16:48:09,131 [NOTICE] - /home/pi/v4l2rtspse
juin 10 16:48:09 raspberrypi v4l2rtspserver[338]:         Play this stream using the URL "rtsp://192.168.1.XXX/unicast```

When I try the last command in VlC like : 
```cvlc rtsp://192.168.1.104 ```
I got :
```
VLC media player 3.0.11.1 Vetinari (revision 3.0.11.1-0-g52483f3ca2)
[0000558d4c855f50] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0000558d4c870730] dummy interface: using the dummy interface module...
Created new TCP socket 7 for connection
Created new TCP socket 6 for connection
[00007f4190001120] live555 demux error: Failed to connect with rtsp://192.168.1.104:554
[00007f4190004950] main stream error: connection failed: Connection refused
[00007f4190004950] satip stream error: Failed to connect to RTSP server 192.168.1.104:554
[00007f4190004950] main stream error: connection failed: Connection refused
[00007f4190004950] access_realrtsp stream error: cannot connect to 192.168.1.104:554
[00007f4190004950] main stream error: Connection failed
[00007f4190004950] main stream error: VLC could not connect to "192.168.1.104:554".
[00007f4198000c80] main input error: Your input can't be opened
[00007f4198000c80] main input error: VLC is unable to open the MRL 'rtsp://192.168.1.104'. Check the log for details.
 ```
Where do I have to look  to resolve this bug ?
Many thanks for this greatfull app.

aweussom, 2021-06-11:
Looks like your server runs on port 8888 - you need to specify port for
cvlc

(VLC uses 554 as default)

hedjour, 2021-06-11:
Your right but when I have corrected this mistake I got :
```
 cvlc rtsp://192.168.1.104:8888/unicast
VLC media player 3.0.11.1 Vetinari (revision 3.0.11.1-0-g52483f3ca2)
[000055e0b2bb5f50] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[000055e0b2bea8f0] dummy interface: using the dummy interface module...
Created new TCP socket 6 for connection
Created new TCP socket 6 for connection
[00007f1450000c00] live555 demux error: no data received in 10s, aborting
```

I will at least use the flux in motion eyes wich say : "unable to open video device"...

aweussom, 2021-06-11:
AFICR, there is no audio for this setup (Streaming a Raspberry Pi camera)

It would be better if the server created a “dummy” mp3 audio track, since
one could then forward to for instance YouTube, which REQUIRES an audio
track in the stream.

hedjour, 2021-06-11:
The raspberry pi do not have access to Internet. Is it possible to disable sound support ?
The system works well with a rpi with internet access I copy the card on another change the ip adress and the rpi refuse all connexion rstp.

mpromonet, 2021-06-11:
Hi,

Based on the command you used to start (using `-P8888`), you should be able to reach the stream with `cvlc rtsp://192.168.1.104:8888/unicast`.

Best Regards,
Michel.

hedjour, 2021-06-16:
Hi,
I have try to edit the /usr/lib/systemd/system/v4l2rtspserver.service like this :
```
[Unit]
Description=V4L2 RTSP server
After=network.target

[Service]
Type=simple
Restart=always
RestartSec=5
ExecStart=/usr/local/bin/v4l2rtspserver -H 1200 -W 1600 -F 60 -P 8888 /dev/video0,
WorkingDirectory=/usr/local/share/v4l2rtspserver

[Install]
WantedBy=multi-user.target
```
Without coma all works at a first time like this :

```bash
systemctl status v4l2rtspserver.service 
Warning: The unit file, source configuration file or drop-ins of v4l2rtspserver.service changed on disk. Run 'systemctl daemon-reload' to reload units.
● v4l2rtspserver.service - V4L2 RTSP server
   Loaded: loaded (/lib/systemd/system/v4l2rtspserver.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-05-22 10:34:29 CEST; 3h 48min ago
 Main PID: 410 (v4l2rtspserver)
    Tasks: 2 (limit: 1938)
   CGroup: /system.slice/v4l2rtspserver.service
           └─410 /usr/local/bin/v4l2rtspserver -H 1200 -W 1600 -F 60 -P 8888 /dev/video0

mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]: 2021-05-22 10:34:30,038 [NOTICE] - /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]:         Device /dev/video0 nb buffer:10
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]: 2021-05-22 10:34:30,650 [NOTICE] - /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:61
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]:         Create Source .../dev/video0
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]: 2021-05-22 10:34:30,652 [NOTICE] - /home/pi/v4l2rtspserver/inc/ServerMediaSubsession.h:47
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]:         format:video/H264
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]: 2021-05-22 10:34:30,652 [NOTICE] - /home/pi/v4l2rtspserver/src/DeviceSource.cpp:93
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]:         begin thread
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]: 2021-05-22 10:34:30,652 [NOTICE] - /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:81
mai 22 10:34:30 pbsup136-76 v4l2rtspserver[410]:         Play this stream using the URL "rtsp://10.16.136.76:8888/unicast"
```
After reload and restart I get this message :

```
systemctl status v4l2rtspserver.service 
● v4l2rtspserver.service - V4L2 RTSP server
   Loaded: loaded (/lib/systemd/system/v4l2rtspserver.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) since Sat 2021-05-22 14:24:59 CEST; 3s ago
  Process: 749 ExecStart=/usr/local/bin/v4l2rtspserver -H 1200 -W 1600 -F 60 -P 8888 /dev/video0, (code=exited, status=0/SUCCESS)
 Main PID: 749 (code=exited, status=0/SUCCESS)
``` 
And no more camera. Have I made something wrong ?

closed 2024-03-24 by mpromonet