mpromonet/v4l2rtspserver #252: RTSP over UDP

vikram-hacklab, 2021-07-12:
Hello friends,

I have been running v4l2rtsp server for over a month now on raspberry pi zero w. It runs smoothly most of the time but sometimes due to network congestion, I start loosing packets and stream becomes very unstable. I am guessing that the stream is currently running over TCP. I understand that the server supports sending data over UDP as well. Can someone guide me on how I can move the stream from TCP to UDP? 

Thanks in advance!!

mpromonet, 2021-07-12:
Hi,

It seems your are swapping UDP & TCP, when there is data congestion UDP packets are dropped, TCP packets are reemitted.

By default the server is sending RTP/UDP traffic, in order to get TCP traffic, you can ask for this sending the RTSP SETUP, setting Transport to `RTP/AVP/TCP`.

Best Regards,
Michel.

aweussom, 2021-07-12:
It’s rather unlikely to be network congestion.
It is VERY likely to the the crappy WiFi in the Zero W.
Stick to UDP - possibly add keyframes more often?

man. 12. jul. 2021 kl. 12:53 skrev vikram-hacklab ***@***.***

vikram-hacklab, 2021-07-14:
> Hi,
>
> It seems your are swapping UDP & TCP, when there is data congestion UDP packets are dropped, TCP packets are reemitted.
>
> By default the server is sending RTP/UDP traffic, in order to get TCP [...]
@mpromonet  Thank you! I guess I figured out threducingreducingreducingreducingreducingreducingreducinge issue. The packets were getting dropped when too many neighbourhood WiFis were streaming data at the same time. Also thank you for the clarification. I had wrongly assumed that the data was transmitted over TCP, by default.  

@aweussom I tried changing the keyframes. It did improve things a bit but not too much. Can I modify anything else to improve the stream quality?

closed 2022-05-10 by mpromonet