mpromonet/v4l2rtspserver #255: build assistance for nix package - linker errors

colemickens, 2021-09-29:
Hi, I am writing a Nix package for this application, as I'd really like to use this - it looks great! thanks!

I do need some help with cmake and building `v4l2rtspserver`. I'm getting close:

```
cmake: enabled parallel building
@nix { "action": "setPhase", "phase": "buildPhase" }
building
build flags: -j12 -l12 SHELL=/nix/store/qfb4j7w2fjjq953nd9xncz5mymj5n0kb-bash-5.1-p8/bin/bash
[  4%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Access.cpp.o
[  9%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Device.cpp.o
[ 13%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Capture.cpp.o
[ 18%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 22%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/DeviceSource.cpp.o
[ 31%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2MmapDevice.cpp.o
[ 31%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Output.cpp.o
[ 40%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o
[ 40%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2ReadWriteDevice.cpp.o
[ 45%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 50%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/logger.cpp.o
[ 54%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MJPEGVideoSource.cpp.o
[ 59%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o
[ 63%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MemoryBufferSink.cpp.o
[ 68%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o
[ 72%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/TSServerMediaSubsession.cpp.o
[ 77%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
/build/source/src/DeviceSource.cpp: In member function 'int V4L2DeviceSource::getNextFrame()':
/build/source/src/DeviceSource.cpp:218:9: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)' declared with attribute 'warn_unused_result' [-Wunused-result]
  218 |    write(m_outfd, buffer, frameSize);
      |    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 81%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o
[ 86%] Linking CXX static library libv4l2wrapper.a
[ 86%] Built target v4l2wrapper
[ 90%] Linking CXX static library liblibv4l2rtspserver.a
[ 90%] Built target libv4l2rtspserver
[ 95%] Building CXX object CMakeFiles/v4l2rtspserver.dir/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(HTTPServer.cpp.o): in function `HTTPServer::HTTPClientConnection::streamSource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
HTTPServer.cpp:(.text+0x1b2): undefined reference to `ByteStreamMemoryBufferSource::createNew(UsageEnvironment&, unsigned char*, unsigned long, unsigned char, unsigned int, unsigned int)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createRtpSink(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0xab): undefined reference to `StreamReplicator::createStreamReplica()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0x258): undefined reference to `PassiveServerMediaSubsession::PassiveServerMediaSubsession(RTPSink&, RTCPInstance*)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::sdpLines()':
MulticastServerMediaSubsession.cpp:(.text+0x308): undefined reference to `PassiveServerMediaSubsession::sdpLines()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD2Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x47): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD0Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x46): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*) [clone .cold]':
MulticastServerMediaSubsession.cpp:(.text.unlikely+0x24): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTI30MulticastServerMediaSubsession[_ZTI30MulticastServerMediaSubsession]+0x18): undefined reference to `typeinfo for PassiveServerMediaSubsession'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0x60): undefined reference to `PassiveServerMediaSubsession::getStreamParameters(unsigned int, unsigned int, Port const&, Port const&, int, unsigned char, unsigned char, unsigned int&, unsigned char&, unsigned char&, Port&, Port&, void*&)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0x68): undefined reference to `PassiveServerMediaSubsession::startStream(unsigned int, void*, void (*)(void*), void*, unsigned short&, unsigned int&, void (*)(void*, unsigned char), void*)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0x98): undefined reference to `PassiveServerMediaSubsession::getCurrentNPT(void*)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0xa8): undefined reference to `PassiveServerMediaSubsession::getRTPSinkandRTCP(void*, RTPSink const*&, RTCPInstance const*&)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0xb0): undefined reference to `PassiveServerMediaSubsession::deleteStream(unsigned int, void*&)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o):(.data.rel.ro._ZTV30MulticastServerMediaSubsession[_ZTV30MulticastServerMediaSubsession]+0xd0): undefined reference to `PassiveServerMediaSubsession::rtcpIsMuxed()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(ServerMediaSubsession.cpp.o): in function `BaseServerMediaSubsession::createSource(UsageEnvironment&, FramedSource*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
ServerMediaSubsession.cpp:(.text+0xe4): undefined reference to `JPEGVideoSource::JPEGVideoSource(UsageEnvironment&)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(ServerMediaSubsession.cpp.o): in function `BaseServerMediaSubsession::createSink(UsageEnvironment&, Groupsock*, unsigned char, std::__cxx11 [... cut ...]
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.data.rel.ro._ZTV16MJPEGVideoSource[_ZTV16MJPEGVideoSource]+0xa0): undefined reference to `JPEGVideoSource::isJPEGVideoSource() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:105: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

Any suggestions?

I'm wondering if `live555` doesn't have correct pkg-config config and thus the linker isn't finding what it needs to link against?

colemickens, 2021-09-29:
I updated our `live555` library to the latest released and my errors look very similar:

```
cmake: enabled parallel building
@nix { "action": "setPhase", "phase": "buildPhase" }
building
build flags: -j12 -l12 SHELL=/nix/store/qfb4j7w2fjjq953nd9xncz5mymj5n0kb-bash-5.1-p8/bin/bash
[  9%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Access.cpp.o
[  9%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 18%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 18%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/DeviceSource.cpp.o
[ 27%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Device.cpp.o
[ 27%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Capture.cpp.o
[ 40%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o
[ 40%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MJPEGVideoSource.cpp.o
[ 40%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2MmapDevice.cpp.o
[ 45%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Output.cpp.o
[ 50%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MemoryBufferSink.cpp.o
[ 54%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o
[ 59%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2ReadWriteDevice.cpp.o
[ 63%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/logger.cpp.o
[ 68%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o
/build/source/src/DeviceSource.cpp: In member function 'int V4L2DeviceSource::getNextFrame()':
/build/source/src/DeviceSource.cpp:218:9: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)' declared with attribute 'warn_unused_result' [-Wunused-result]
  218 |    write(m_outfd, buffer, frameSize);
      |    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 72%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/TSServerMediaSubsession.cpp.o
[ 77%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
[ 81%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o
[ 86%] Linking CXX static library libv4l2wrapper.a
[ 86%] Built target v4l2wrapper
[ 90%] Linking CXX static library liblibv4l2rtspserver.a
[ 90%] Built target libv4l2rtspserver
[ 95%] Building CXX object CMakeFiles/v4l2rtspserver.dir/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(HTTPServer.cpp.o): in function `HTTPServer::HTTPClientConnection::streamSource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
HTTPServer.cpp:(.text+0x1b2): undefined reference to `ByteStreamMemoryBufferSource::createNew(UsageEnvironment&, unsigned char*, unsigned long, unsigned char, unsigned int, unsigned int)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createRtpSink(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0xab): undefined reference to `StreamReplicator::createStreamReplica()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0x258): undefined reference to `PassiveServerMediaSubsession::PassiveServerMediaSubsession(RTPSink&, RTCPInstance*)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::sdpLines()':
MulticastServerMediaSubsession.cpp:(.text+0x308): undefined reference to `PassiveServerMediaSubsession::sdpLines()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD2Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x47): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD0Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x46): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*) [clone .cold]':
MulticastServerMediaSubsession.cpp:(.text.unlikely+0x24): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
[... cut ...]
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MJPEGVideoSource.cpp.o):(.data.rel.ro._ZTV16MJPEGVideoSource[_ZTV16MJPEGVideoSource]+0xa0): undefined reference to `JPEGVideoSource::isJPEGVideoSource() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:105: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

colemickens, 2021-09-29:
Ignore the reply I deleted, this is the new build log after updating `live555` to latest (though, note, we may be building it wrong?)

```
@nix { "action": "setPhase", "phase": "buildPhase" }
building
build flags: -j12 -l12 SHELL=/nix/store/qfb4j7w2fjjq953nd9xncz5mymj5n0kb-bash-5.1-p8/bin/bash
[  9%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Capture.cpp.o
[ 13%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Access.cpp.o
[ 13%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Device.cpp.o
[ 22%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/H264_V4l2DeviceSource.cpp.o
[ 22%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/DeviceSource.cpp.o
[ 36%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ALSACapture.cpp.o
[ 36%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2MmapDevice.cpp.o
[ 36%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/HTTPServer.cpp.o
[ 40%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2Output.cpp.o
[ 45%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/V4l2ReadWriteDevice.cpp.o
[ 50%] Building CXX object CMakeFiles/v4l2wrapper.dir/v4l2wrapper/src/logger.cpp.o
[ 54%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MJPEGVideoSource.cpp.o
[ 59%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MemoryBufferSink.cpp.o
[ 63%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MulticastServerMediaSubsession.cpp.o
[ 68%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/ServerMediaSubsession.cpp.o
[ 72%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/TSServerMediaSubsession.cpp.o
[ 77%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
[ 81%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o
/build/source/src/DeviceSource.cpp: In member function 'int V4L2DeviceSource::getNextFrame()':
/build/source/src/DeviceSource.cpp:218:9: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)' declared with attribute 'warn_unused_result' [-Wunused-result]
  218 |    write(m_outfd, buffer, frameSize);
      |    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 86%] Linking CXX static library libv4l2wrapper.a
[ 86%] Built target v4l2wrapper
[ 90%] Linking CXX static library liblibv4l2rtspserver.a
[ 90%] Built target libv4l2rtspserver
[ 95%] Building CXX object CMakeFiles/v4l2rtspserver.dir/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(HTTPServer.cpp.o): in function `HTTPServer::HTTPClientConnection::streamSource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
HTTPServer.cpp:(.text+0x1c2): undefined reference to `ByteStreamMemoryBufferSource::createNew(UsageEnvironment&, unsigned char*, unsigned long, unsigned char, unsigned int, unsigned int)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::sdpLines()':
MulticastServerMediaSubsession.cpp:(.text+0x8d): undefined reference to `PassiveServerMediaSubsession::sdpLines(int)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createRtpSink(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0x1bb): undefined reference to `StreamReplicator::createStreamReplica()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*)':
MulticastServerMediaSubsession.cpp:(.text+0x378): undefined reference to `PassiveServerMediaSubsession::PassiveServerMediaSubsession(RTPSink&, RTCPInstance*)'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD2Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x47): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::~MulticastServerMediaSubsession()':
MulticastServerMediaSubsession.cpp:(.text._ZN30MulticastServerMediaSubsessionD0Ev[_ZN30MulticastServerMediaSubsessionD5Ev]+0x46): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: liblibv4l2rtspserver.a(MulticastServerMediaSubsession.cpp.o): in function `MulticastServerMediaSubsession::createNew(UsageEnvironment&, in_addr, Port, Port, int, StreamReplicator*) [clone .cold]':
MulticastServerMediaSubsession.cpp:(.text.unlikely+0x24): undefined reference to `PassiveServerMediaSubsession::~PassiveServerMediaSubsession()'
[... cut ...]
/nix/store/fp2h123793bxln4cjxy0mz08sf4nx2m3-binutils-2.35.1/bin/ld: /nix/store/58pxli8qj262mk75hr40zbmd4pdyd7fz-live555-2021.08.24/lib/libliveMedia.a(HMAC_SHA1.o): in function `sha1(unsigned char*, unsigned char const*, unsigned int, unsigned char const*, unsigned int)':
(.text+0x73): undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:105: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

colemickens, 2021-09-30:
I had to:
1. patch `live555` to output decent pkg-config data
2. patch this app to find openssl+live555 via pkg-config
3. patch this app to find VERSION from a passed in cmake variable

This has made it possible to package it in a relatively straightforward way for NixOS:
https://github.com/cole-mickens/nixcfg/blob/8af67fa760bae7f8c9a2e2b1dce04634d6da76e0/pkgs/v4l2rtspserver/default.nix

mpromonet, 2021-09-30:
Hi,

I know this is a bit confusing, my original idea was to build live555 in order o set the CFLAGS, and PR propose to use pkg-tool to use it.  There is some imporovement to do. openssl is needed only without `NO_OPENSSL=1`.

Feel free to propose some using PR, if it helps you and maybe others users.

Best Regards,
Michel.