mpromonet/v4l2rtspserver #256: package architecture (amd64) does not match system (arm64)

andrea0000, 2021-11-06:
The Debian package for arm64 cannot be installed.

Below is the output of the installation attempt:

pi@raspberrypi:~$ sudo apt install ./v4l2rtspserver-0.2.3-Linux-arm64.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'v4l2rtspserver:amd64' instead of './v4l2rtspserver-0.2.3-Linux-arm64.deb'
The following NEW packages will be installed:
  v4l2rtspserver:amd64
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/505 kB of archives.
After this operation, 1,787 kB of additional disk space will be used.
Get:1 /home/pi/v4l2rtspserver-0.2.3-Linux-arm64.deb v4l2rtspserver amd64 0.2.3 [505 kB]
dpkg: error processing archive /home/pi/v4l2rtspserver-0.2.3-Linux-arm64.deb (--unpack):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 /home/pi/v4l2rtspserver-0.2.3-Linux-arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)_

I am using the latest version of Raspberry Pi OS for arm64: 
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

mpromonet, 2021-11-06:
Hi @andrea0000 

It should be fixed with https://github.com/mpromonet/v4l2rtspserver/releases/download/v0.2.4/v4l2rtspserver-0.2.4-Linux-arm64.deb

Best Regards,
Michel.

andrea0000, 2021-11-06:
Now I can install the package.
Thanks.

closed 2021-11-06 by mpromonet