mpromonet/v4l2rtspserver #258: Unable to capture stdout when running in /etc/init.d

tachang, 2021-11-14:
**Describe the bug**

When I run the following command the output is as expected. Stdout is sent to the file and stderr is captured.

`/usr/bin/v4l2rtspserver $AUTH_PARAMS -A 48000 -C 1 -a S16_LE -l 1 /dev/video3,hw:0,1 /dev/video5 1> /var/log/v4l2rtspserver-stdout.log 2> /var/log/v4l2rtspserver-stderr.log &`

However when the same command is run in the etc/init.d system on startup I only get for stdout this:

```
# cat v4l2rtspserver-stdout.log 
log level:500
# 
```

I don't see any of the [NOTICE] log messages. Anyone have any ideas why this is? I do roughly the same thing for other processes and it seems to work. Is there something special about the logging in this project that causes this to happen?

mpromonet, 2021-11-14:
Hi,

Increasing the log level may help you ? Maybe some sound access without login session ?

Best Regards,
Michel.

tachang, 2021-11-15:
Odd. I think log level is fine. It seems like when I background the process using the ampersand `&` it fails to log to stdout.

tachang, 2021-11-15:
I also get this behavior when I pipe it to tee.
```
# /usr/bin/v4l2rtspserver -P 8555 /dev/video3 /dev/video5 | tee
log level:500
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
```

tachang, 2021-11-15:
I wonder if log4cpp is missing the ImmediateFlush parameter so I am seeing it all getting buffered.

mpromonet, 2021-11-16:
Hi,

To unbuferred stdout you may look to https://unix.stackexchange.com/questions/25372/turn-off-buffering-in-pipe

Best Regards,
Michel.

closed 2022-02-06 by mpromonet