mpromonet/v4l2rtspserver #261: Setting stream parameters

dbrb2, 2021-12-29:
This isn't a bug, so much as a request for clarification: 

I can specify frame rate and resolution for an H.264 video source (in this case a pi camera with the bcm2835-v4l2 driver) as follows: 
v4l2rtspserver -F $FPS -H $Y -W $X /dev/video0

I can also set some other stream parameters using v4l2-ctrl at the terminal - such as: 
v4l2-ctl -d /dev/video0 -c video_bitrate=4000000

On paper, I can also set frame rate at the terminal (when the device is not in use) - v4l2-ctl -d /dev/video0 -p 10

However, if I do that, and then subsequently start v4l2rtspserver without a "-F" parameter, the frame rate defaults to 25fps 

Does v4l2rtspserver set frame rate and resolution to some default if not specified - overriding any preceding v4l2-ctl commands?

mpromonet, 2021-12-30:
Hi,

You are right, default settings is 25fps, this could be better to not set fps by default.
There is 2 possible ways either using '-f ' (no ioctl will be send and then current capture settings will be used), or using '-F0' (fps will not be set).

Best Regards,
Michel

dbrb2, 2021-12-30:
Ah thanks. 
Setting just -f still seemed to override the configured frame rate with 25fps 
However, setting -F0 left it as set 

Interestingly, whilst leaving out an explicit F defaulted to 25fps, not specifying W and H does not seem to result in these being overridden - they stay set to whatever their preconfigured valued were 

Cheers!

dsmahalle, 2022-04-15:
can I add timestamp overlay to video

mpromonet, 2022-04-15:
Hi,

If your device support it, see related question https://github.com/mpromonet/v4l2rtspserver/issues/93

Best Regards,
Michel.