mpromonet/v4l2rtspserver #262: v4l2 un support raw video format

dcboy, 2021-12-30:
**Describe the bug**
```
rk3288:/data/local/tmp # ./v4l2rtspserver /dev/video0
log level:500

[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/main.cpp:294
	Version: 0.2.4-8-g79a13bd-dirty live555 version:2021.12.18
[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video0
[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:uvcvideo capabilities:84200001 mandatory:4000001
[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video0 support streaming
[ERROR] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:H264 format is:YUYV
[ERROR] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:MJPG format is:YUYV
[ERROR] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:JPEG format is:YUYV
[ERROR] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:83
	Cannot init device:/dev/video0
[NOTICE] /Users/dcboy/D/github/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: android
 - OS: android 7.1
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

mpromonet, 2021-12-30:
Hi,

You may try '-fYUYV'

Best Regards,
Michel

dcboy, 2022-01-05:
Thx that ok， that h264 encoder is using hardware？

mpromonet, 2022-01-05:
Hi,

This program does not encode, it streams over RTSP v4L2 capture frames with supported format (H264, H265, VP8, VP9, JPEG, YUYV is a quick try).

Best Regards,
Michel.

closed 2022-02-06 by mpromonet

abhinovpankaj, 2024-09-14:
Doesn;t work. 
vicedeck@e3cam:~/v4l2rtspserver $ v4l2rtspserver -f YUYV /dev/video0
log level:500

[NOTICE] (main.cpp:249)                	Version: 0.3.11-4-g410c0d4 live555 version:2020.01.19
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
[NOTICE] (V4l2RTSPServer.cpp:37)       	Create V4L2 Source...YUYV
[NOTICE] (V4l2RTSPServer.cpp:37)       	Create V4L2 Source.../dev/video0
VIDIOC_STREAMON: Invalid argument
[WARN]   (V4l2Device.cpp:243)          	Cannot set param for device:/dev/video0 Inappropriate ioctl for devicedeck@e3cam:~/v4l2rtspserver $

mpromonet, 2024-09-14:
@abhinovpankaj 

It seems /dev/video0 is not a video device, you might try v4l2-utils...

Best Regards 
Michel