mpromonet/v4l2rtspserver #263: FR: get an image/snapshot during RTSP streaming

Ph1975, 2022-01-01:
I use a raspberry as a door-bell with camera (DoorPi). 
During my absence i´d like to get a Snapshot of the camera as soon as someone rings the door. The v4l2rtspserver blocks the camera (busy device) so i cannot use any of the standard tools to get a quick snapshot.

On the other hand using ffmpeg to grab a snapshot from the stream takes quite some time until the image is pulled/converted/saved - so the visitor already might be out of sight of the camera (with my installation it is around 8-10 seconds from sending the shell-command until the file is stored).

It would be great to have a possibility to get a fast snapshot (2-3 seconds) out of the stream directly through the server.

I use a raspi 3, GBit LAN

ricardoboss, 2022-03-16:
I'm having a similar use-case with octoprint, where I want to be able to stream the webcam feed as well as take regular snapshots to create a time lapse.

mpromonet, 2022-03-16:
Hi @ricardoboss 

Please read the [CONTRIBUTING.md](https://github.com/mpromonet/v4l2rtspserver/blob/master/.github/CONTRIBUTING.md), it is really short.

Best Regards,
Michel.