mpromonet/v4l2rtspserver #265: Green screen when using MJPG and non default resolution

FBITuco, 2022-01-07:
**Describe the bug**
A clear and concise description of what the bug is.
Hello.
I found similar problem as in #170 and #140 
When using camera "default" resolution 1920x1080 (As default i mean first resolution listed in "v4l2-ctl -d /dev/video0 --list-formats-ext"), all works ok. Used command is: v4l2rtspserver -vvv -fMJPG -W 1920 -H 1080 -A 8000 -C 1 -a S16_LE /dev/video0,pcm.mic_lef
t
See -vvv log from v4l2rtspserver: 
[v4l2rtspserver-1080.log](https://github.com/mpromonet/v4l2rtspserver/files/7827407/v4l2rtspserver-1080.log)

See log from ffplay: 
[ffplay-1080.log](https://github.com/mpromonet/v4l2rtspserver/files/7827408/ffplay-1080.log)

But when I try to use different resolution, I only can see green screen with some different colors in top. Command example: v4l2rtspserver -vvv -fMJPG -W 1280 -H 720 -A 8000 -C 1 -a S16_LE /dev/video0,pcm.mic_left
[image]

See -vvv log from v4l2rtspserver: 
[v4l2rtspserver-720.log](https://github.com/mpromonet/v4l2rtspserver/files/7827410/v4l2rtspserver-720.log)

See log from ffplay: 
[ffplay-720.log](https://github.com/mpromonet/v4l2rtspserver/files/7827411/ffplay-720.log)

**Desktop (please complete the following information):**
 - OS: OpenWRT
 - Version 0.2.4, live555 2021.10.28
 - Video:
 v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1920x1080
			Interval: Discrete 0.111s (9.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.200s (5.000 fps)

Thanks for your work!

khw1006, 2022-09-11:
Hello, [FBITuco](https://github.com/FBITuco).
Did you solve this problem? I have a same problem.
And where did you get 0.2.4 version? I can only find 0.2.1 version.

mpromonet, 2022-09-11:
Hi @khw1006,

Last release is 0.3.3, you can get it from https://github.com/mpromonet/v4l2rtspserver/releases

Best Regards,
Michel.

FBITuco, 2022-09-14:
Hello @khw1006.
I believe that this is not bug, it is a "feature".
RTSP will not transfer huffman tables. RTSP receiver is using tables defined in JPEG standard (See [https://ffmpeg.org/doxygen/trunk/jpegtables_8c_source.html](url).
Some webcams will use different huffman tables for different resolutions. This is why when you capture .jpeg image from camera and open it, all looks ok, but images sended over RTSP will not work.

There is no simple way how to "fix" this. You can modify rtspserver application (Need to convert received .jpeg from camera to image with "correct" huffman tables. Captured frame from camera is here: [https://github.com/mpromonet/v4l2rtspserver/blob/master/src/V4L2DeviceSource.cpp#L168](url)), or you can modify application which is receiving stream with huffman tables corresponding to .jpeg image from camera.

Or You need to buy different camera :-).
Have a nice day!

This issue probably can be closed.

closed 2023-04-29 by mpromonet